PX4/PX4-Autopilot #2865: DX8 Aux2/Channel 8

meriachee, 2015-09-15:
The Aux3 output from my DX8 does not register on the Pixhawk as channel 8. Aux2 works fine as channel 7 and all of the other functions work great. I see the state changing on the DX8, and I do not see the state changing on the Mav screen. Any ideas?

LorenzMeier, 2015-09-15:
Are you sure you have an 8 channel receiver?

meriachee, 2015-09-15:
It's a DX8, bound to a Spektrum satellite receiver, as per the PixHawk instructions. The Tarot flys as expected, Aux 2 is built as indicated in the wiki and works just fine, but Aux3 will not. The settings and switches can be swapped between Aux2 and Aux3 and there is no response from Aux3. The DX8 shows the change in switch states in the lcd as working.

LorenzMeier, 2015-09-15:
The fact that the RC shows that channel working doesn't mean anything. Please re-confirm that you have in fact an 8 channel receiver. Spektrum defaults for many kits to 7:
https://www.spektrumrc.com/Products/Default.aspx?ProdId=SPM6070

meriachee, 2015-09-15:
The Satellite receiver is
https://www.spektrumrc.com/Products/Default.aspx?ProdID=SPM9645
It is directly connected to the SPM port on the Pixhawk. I used an AR6210 to bind it. Are you saying that the 9645 receiver is limited or defaulted in the channels it has available by the receiver used to bind it?

LorenzMeier, 2015-09-15:
Yes. Could you try to either bind it through QGroundControl or bind it using a Spektrum receiver having more than 7 channels? I have personally tested that we support the maximum channel count.

meriachee, 2015-09-15:
Interesting (thank you by the way)  A larger receiver is unnecessary if the software will do it, so I will try this right away.

meriachee, 2015-09-15:
Ok, now I feel really dumb. In the QGroundControl program, version 2.7.1 I cannot find the radio page that the Pixhawk configuration page references.
https://pixhawk.org/peripherals/radio-control/spektrum_dsm

LorenzMeier, 2015-09-20:
@meriachee The equivalent UI is available in version 2.7.1. It just doesn't look 1:1 the same. Let us know if you feel something is indeed missing.

closed 2015-09-20 by LorenzMeier