PX4/PX4-Autopilot #2893: [TECS] Pitch down when engaging alt ctrl for first time

mswingtra, 2015-09-23:
During flight test, when altitude position control has been engaged airplane started to pitch down. The pitch integral reached high negative value at that time. We flew with master 3724947 and we used TECS fix https://github.com/PX4/Firmware/pull/2783
[image]

Logging: http://logs.uaventure.com/view/Mz4PbA4F32q9ekaH85xudP

tumbili, 2015-09-23:
@mswingtra Are you sure you are using the right version? We are logging the pitch and the thrust integrator for TECS but I can't see it in LogMuncher?

mswingtra, 2015-09-23:
It seems that there has changed something in logging upstream, and since we update roughly once a month we did not have that change this time. We will have it by the next flight, so then we can provide more data. For now, either this issue can be kept open and we'll update it ASAP, or we'll reopen it in two-three weeks.

AndreasAntener, 2015-10-01:
@tumbili we noticed a TECS init issue today, with latest master. We did 2 flights, when POSCTL was engaged the first time we had a huge pitch integrator:
[image]
[image]
Log: http://logs.uaventure.com/view/oFMJ6Etfx2XsZA6zdLkypc

tumbili, 2015-10-01:
@AndreasAntener Thanks for reporting! Seems like this issue is still present after all.

LorenzMeier, 2015-10-04:
This here might be another instance of it:
http://logs.uaventure.com/view/CRzMpDoUvSshcFPY8wNXUA

LorenzMeier, 2015-11-01:
So we should really have fixed this now. Patch is on master.

closed 2015-11-01 by LorenzMeier