PX4/PX4-Autopilot #2898: uORB pub/sub regression

LorenzMeier, 2015-09-23:
@mcharleb It is possible that we have a regression in muORB. We have user reports on NuttX and I have severe stability issues on Mac OS.

This branch will help to reproduce:
https://github.com/PX4/Firmware/tree/cmake-2-dbg

It is important that you have a complete setup including gamepad - the main symptom is that sensor values stop updating and the vehicle will drop out of the sky after 1-10 seconds. There is a also this troubling user report on NuttX:

https://groups.google.com/forum/#!topic/px4users/tP0kz9XPSjE

LorenzMeier, 2015-09-23:
The crash on Mac OS happens because _pollset[i] has an invalid entry, leading to a crash when trying to post the semaphore: https://github.com/PX4/Firmware/blob/cmake-2/src/drivers/device/vdev.cpp#L416

LorenzMeier, 2015-09-23:
The 2nd symptom is that without the usleeps on startup the system can get into an infinite loop where it stops updating the sensors.

rkintada, 2015-09-24:
Just to clarify, muorb will not be active for nuttx or macos builds as it is specific to snapdragon based platform.  I am suspecting this is more specific for uorb stack.  Also there are some unit tests for muorb related to changes to uorb stack.

LorenzMeier, 2015-11-25:
Fixed.

closed 2015-11-25 by LorenzMeier