PX4/PX4-Autopilot #2911: INAV [UAVCAN?] GPS timeout

mhkabir, 2015-09-25:
I tried getting some sort of a stable/flyable/reliable setup up today which wouldn't need debugging, flashed master to a fresh Pixhawk, running a Zubax GNSS over UAVCAN.
I currently keep getting messages of ```[inav] gps timeout``` spamming the shell and over mavlink. This seemed to be the behavior that occurs if another driver publishing on the same topic, but I stopped the native GPS driver, and the issue still persists. I think this is a regression introduced in the new UAVCAN updates. Could someone please confirm that they can reproduce this behavior (@Aperture-Laboratories @AndreasAntener @SimonWilks ? 
And while we're at it, could we maybe setup some form of a real test rig to test every commit against UAVCAN on a Pixhawk? Because honestly, I'm ready to spend *some* debugging time getting things to work, but this is getting frustrating to have have issues continuously with one thing or the other. I would probably now prefer to settle for classic serial GPS considering the amount of time sunk trying to get everything working.
On a side note, the UAVCAN external compass use seems to be malfunctioning as well, probably because of the CAN issue. I'm not sure how I can check what exactly is wrong, but the external compass is being ignored again.

pavel-kirienko, 2015-09-25:
Hi Kabir, could you describe please your configuration, namely the value of `SYS_AUTOSTART`, whether you're using any additional init scripts, and the exact commit on master which triggers your timeouts. Thanks.

mhkabir, 2015-09-26:
SYS_AUTOSTART - 10019
No additional scripts, just verbatim master @ 05a08252ff3f29b5ac684e97531a26a09e1310a9

pavel-kirienko, 2015-09-26:
Can confirm, there was a regression in master. I'm looking into it.

pavel-kirienko, 2015-09-26:
@LorenzMeier any chance we could extend the existing hardware testing suite with UAVCAN peripherals?

LorenzMeier, 2015-09-26:
Yes, that's certainly possible. But more importantly we need an active and invested maintainer.

mhkabir, 2015-09-26:
I'd like to be an active and interested maintainer :) Seeing as I'm
sinking enough time into this already, I'd be one of the happiest to
see this working out.

LorenzMeier, 2015-09-26:
Hi Kabir,

A maintainer needs to go in, debug and fix things as well as providing tests to ensure quality. You have been a great tester and raised a lot of issues, but I'd like to encourage you to make the last step and move to provide fixes.

Thanks!

pavel-kirienko, 2015-09-26:
Bisection shows that the regression appeared with https://github.com/PX4/Firmware/pull/2895. I wasn't able to find the root cause yet.

FYI @davids5

pavel-kirienko, 2015-09-26:
https://github.com/PX4/Firmware/pull/2914

mhkabir, 2015-10-10:
Closing this, since it's actually #2915 now. And timeouts are fixed by the current hacky state.

closed 2015-10-10 by mhkabir

pavel-kirienko, 2015-10-10:
Yesp, this one has actually been resolved with #2914.