PX4/PX4-Autopilot #2913: [sitl] compile error

thomasgubler, 2015-09-26:
```
/home/thomasgubler/src/Firmware/src/modules/dataman/dataman.c:717:4: fatal error: unterminated function-like macro invocation
                        warnx("Incompatible data manager file %s, resetting it", k_data_manager_device_path);
                        ^
/home/thomasgubler/src/Firmware/src//modules/systemlib/err.h:81:25: note: expanded from macro 'warnx'
#define warnx(...)                                      PX4_WARN(__VA_ARGS__)
                                                        ^
/home/thomasgubler/src/Firmware/src/platforms/px4_log.h:371:29: note: expanded from macro 'PX4_WARN'
#define PX4_WARN(FMT, ...)      __px4_log_file_and_line(_PX4_LOG_LEVEL_WARN,  FMT, ##__VA_ARGS__)
                                ^
/home/thomasgubler/src/Firmware/src/platforms/px4_log.h:246:2: note: expanded from macro '__px4_log_file_and_line'
        __px4__log_startline(level)\
        ^
/home/thomasgubler/src/Firmware/src/platforms/px4_log.h:146:75: note: expanded from macro '__px4__log_startline'
#define __px4__log_startline(level)     if (level <= __px4_log_level_current) printf(
                                                                              ^
/usr/include/bits/stdio2.h:107:11: note: macro 'printf' defined here
#  define printf(...) \
```

The problem probably stems from the open paranthesis here since printf is a macro:
https://github.com/PX4/Firmware/blob/master/src/platforms/px4_log.h#L146

LorenzMeier, 2015-09-26:
I recommend to build with the specced CLANG version in the readme.

thomasgubler, 2015-09-26:
I expected the answer ;-) Still we will need to fix this at some point.
```
clang --version
clang version 3.6.2 (tags/RELEASE_362/final)
```

commit 1ae8483e (2015-09-27): fix 'unterminated function-like macro invocation', fixes #2913
closed 2015-09-27 by thomasgubler