PX4/PX4-Autopilot #2929: Experimental VTOL TBS Cairpiranha

binary42, 2015-09-29:
The autostart ID of 13001 from https://pixhawk.org/platforms/vtol/tbs_caipirinha_vtol#software does not sem to be correct. When modified, b/e flashes after restart and onced arm no esc nore disarm activity is permitted. What is the correct autostart ID for the TBS VTOL airframe configuration?

tumbili, 2015-09-29:
@binary42 This should be the correct ID. Can you give some more information what's exactly happening? What does QGroundControl tell you?

binary42, 2015-09-29:
@tumbili The only indication that I see on qgroundcontrol is in air frame: confirmation that there is a custom air frame in use. I can reset it as well. I've tried the other experimental air frame ids and the same response with the b/e indicator occurs. The ID is set and reported as 13001 as well in the parameters list. I also tied setting the autoconfig ID to 1 as well, but this didn't seem to change behavior.

Generally, I load oup the firmware, from the master branch, set the autostart ID and calibrate. Upon restart of the pixhawk, the b/e starts flashing. I can arm the unit via the safety switch and radio, but thats it. I've tried variations on servo and motor attachment locations on the pixhawk as well. I've also tried setting the ID and restarting prior to calibration, but it made zero difference. 

I've tested the servos and escs independently through the receiver and confirmed they operate as expected.

tumbili, 2015-09-29:
@binary42 Could you send me a log please. Arm the vehicle, move your sticks around a bit and then please send me that log.

tumbili, 2015-09-29:
You can upload it here http://logs.uaventure.com/

binary42, 2015-09-29:
@tumbili, failing to download logs. I'm going to try from scratch for the setup and see if I can grab the logs afterwards. Also, noticing in the files list in /etc/init.d that all the experimental VTOL have an INACTIVE_ file name. Not sure if that means anything for configuration. I also do not see an rc.autostart script as called in config.

binary42, 2015-09-29:
wait, I take that back, there is an rc.autostart

binary42, 2015-09-29:
@tumbili. Thank you for the help, it seems after a fresh reinstall of the firmware from the dev/master branch has fixed the issue. I must have had a corrupted install...or something like that. Anyway, I am now having isues with the escs arming. The b/e indicator now remains off and the servos...twitch. I'm going to go through this and update soon. Thanks so much for your time!

tumbili, 2015-09-29:
@binary42 No problem, please send me the log if you have problems.

binary42, 2015-09-29:
Quick question, what orientation setting do you use?

tumbili, 2015-10-06:
@binary42 Sorry, have been really busy at conferences. Orientation was roll 270, yaw 270 I think. Are you still having issues with the setup or is it working?

binary42, 2015-10-07:
No worries, we set it up in another config and got it running in any event. However, I think we need a bit more work with it. I keep getting battery and radio failsafe alarms during flight when I shouldn't be. Can't elaborate at the moment. We just retuned for our vehicle and will continue testing over the next two weeks. I will let you know how it goes. Thanks again and great job getting a tail sitter bi-rotor in the stack!

closed 2015-11-30 by tumbili