PX4/PX4-Autopilot #2937: Inspect math exception handling path and linkage for CMake NuttX

LorenzMeier, 2015-10-03:
@davids5 The cmake-2 branch needs this to compile:

```
#if 0
#ifdef __cplusplus
__EXPORT int matherr(struct __exception *e)
{
	return 1;
}
#else
__EXPORT int matherr(struct exception *e)
{
	return 1;
}
#endif
#endif
```

Which raises a few red flags for me.

LorenzMeier, 2015-10-30:
@davids5 Would be great if you could look into this

davids5, 2015-11-13:
@LorenzMeier 

I suspected the worst. But after analyzing the differences in compiler options from nuttx_next to cmake:

[image]

I think the reason is not so bad, The original make build linked object files against library files. The cmake build for the most part builds libraries and links them (there is only one object file linked and that is  builtin_commands).  This is what is causing the duplicate definition.

As a simple test moving the definition of matherr to the template for builtin_commands.c, the one linked object file, builds just fine.

So Options are:

1) Do nothing - the code from libm returns 0 not 1 as our stub does
```
      If the matherr() function returns zero, then the system sets errno as
       described above, and may print an error message on standard error
       (see below).

       If the matherr() function returns a nonzero value, then the system
       does not set errno, and doesn't print an error message.
```
2) use a -Wl,-wrap matherr - to hide the library version

3) move matherr to linked object module or force the file it is in to linked as an object module.

LorenzMeier, 2015-11-25:
Its fine as is.

closed 2015-11-25 by LorenzMeier