PX4/PX4-Autopilot #2942: multirotor SITL: wrong mag values subscribed

bansiesta, 2015-10-05:
The wrong mag values are selected in the voting in the attitude_estimator_q in multirotor SITL.

It seems that the priority of MAG 1 is higher than MAG 0, therefore the mag topic 1 instead of 0 is subscribed which is all 0.

I'm not quite sure at what level this should be fixed or worked around. I assume the mag topic 1 has higher priority because it generally refers to the external mag? Where is this sensor priority (respectively uORB) priority encoded?

```
WARNING: degenerate accel / mag! (file Firmware/src/modules/attitude_estimator_q/attitude_estimator_q_main.cpp line 334)
```

LorenzMeier, 2015-10-05:
In the orb_advertise_multi call in src/modules/simulator

We probably should add the zero check to the validator itself, independent of that the sim needs to be fixed.

I also want to add a sim run to CI.

LorenzMeier, 2015-10-05:
The true cause is that SITL has no noise and the same value check triggers.

LorenzMeier, 2015-10-05:
Fixed in https://github.com/PX4/jMAVSim/commit/e767ec93044fb6a9def6d5ed9f5276ded068af3f

closed 2015-10-05 by LorenzMeier

bansiesta, 2015-10-06:
Thanks for fixing this, although it broke my jMAVSim compilation:
https://github.com/PX4/jMAVSim/issues/12

bansiesta, 2015-10-06:
This really gets it working again: https://github.com/PX4/jMAVSim/pull/13

bansiesta, 2015-10-06:
> In the orb_advertise_multi call in src/modules/simulator

And where are the UORB priorites defined? Somehow the hardware priorities should not have an influence on the SITL, right?

LorenzMeier, 2015-10-06:
During publication. SITL obviously should mimic the real system and therefore does publish with different priorities.

LorenzMeier, 2015-10-06:
FYI: I added jMAVSim with Oracle and OpenJDK to Travis:
https://travis-ci.org/PX4/jMAVSim/jobs/83838791