PX4/PX4-Autopilot #2956: SITL with Gazebo, make robust, testing

tumbili, 2015-10-07:
PX4 SITL with gazebo (without the ROS tail) is now in a state where the IRIS actually flies.
See instructions here https://github.com/tumbili/sitl_gazebo how to use it.
The IRIS does not fly very well yet, maybe there is a bug somewhere, maybe it's just tuning, we have to see. Would be great is somebody could try it out, graphics and physics are way better than the ones from jMAVSim :-)
The interface to the PX4 app is exactly the same as with jMAVSim.

tumbili, 2015-10-07:
Thanks to @devbharat  for doing most of the work on this!

devbharat, 2015-10-07:
@tumbili  thanks for the updated readme! Maybe adding a note to use the sitl_quad_gazebo init script (make sitl_quad_gazebo) would be helpful. I think it's tuned a little better, along with some other changes.

jgoppert, 2015-10-07:
Looking forward to trying this out, nice work!

v01d, 2015-10-07:
I tried compiling it and I got this:

/home/v01d/coding/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘virtual void gazebo::GazeboMavlinkInterface::Load(gazebo::physics::ModelPtr, sdf::ElementPtr)’:
/home/v01d/coding/sitl_gazebo/src/gazebo_mavlink_interface.cpp:73:10: error: ambiguous overload for ‘operator=’ (operand types are ‘gazebo::math::Vector3’ and ‘<brace-enclosed initializer list>’)
   mag_W_ = {0.21523, 0.00771, 0.42741};
          ^
In file included from /usr/include/gazebo-6.1/gazebo/common/Color.hh:24:0,
                 from /usr/include/gazebo-6.1/gazebo/common/common.hh:13,
                 from /home/v01d/coding/sitl_gazebo/include/gazebo_mavlink_interface.h:23,
                 from /home/v01d/coding/sitl_gazebo/src/gazebo_mavlink_interface.cpp:22:
/usr/include/gazebo-6.1/gazebo/math/Vector3.hh:178:24: note: candidate: gazebo::math::Vector3& gazebo::math::Vector3::operator=(const Vector3d&)
       public: Vector3 &operator=(const ignition::math::Vector3d &_v);
                        ^
/usr/include/gazebo-6.1/gazebo/math/Vector3.hh:183:24: note: candidate: gazebo::math::Vector3& gazebo::math::Vector3::operator=(const gazebo::math::Vector3&)
       public: Vector3 &operator =(const Vector3 &_v);
                        ^
CMakeFiles/rotors_gazebo_mavlink_interface.dir/build.make:62: recipe for target 'CMakeFiles/rotors_gazebo_mavlink_interface.dir/src/gazebo_mavlink_interface.cpp.o' failed

v01d, 2015-10-07:
Fixed it by changing the assignment to: `... = gazebo::math::Vector3{...}`

tumbili, 2015-10-07:
@v01d Can you try this repo here https://github.com/tumbili/sitl_gazebo/tree/master/src ?

tumbili, 2015-10-07:
It should now use the correct coordinate system (NED). Before it was using another one.

LorenzMeier, 2015-10-07:
@tumbili Could you please work out with @devbharat which version you want to maintain and make sure we're not split across many branches with this? I also need to talk to @devbharat how we want to get this playing nicely with RotorS. The plan was not necessarily to create the next Sim fork, but rather to architect it so that we do not depend on ROS, but can leverage it when its installed.

tumbili, 2015-10-07:
@v01d Could you also double check this commit https://github.com/tumbili/sitl_gazebo/commit/46aed6d02c57ecc04cd03fce5bda04e63716db1c
The simulator uses a coordinate system where z points up, x to the front and y to the left.

v01d, 2015-10-07:
@tumbili I will pull your latest change and test. I'm currently re-compiling gazebo, since it wasnt working.

v01d, 2015-10-07:
@tumbili I just re-build gazebo and then your plugin. I copied the files as instructed in the README (I created a "iris" folder inside the models dir, since I noticed other models where organized like that).
When I run gazebo and click on the iris model, I only get a spinning cursor and nothing happens. Any idea why?

tumbili, 2015-10-07:
@v01d Can you only insert the model after gazebo has been running for 1 minute? We might have to open an issue for this.

devbharat, 2015-10-07:
@v01d   Did you add the gazebo plugin path environment variable ? gazebo is probably having a hard time locating the .so files of the plugins you just generated ...

v01d, 2015-10-07:
Now it worked (partially). It seems I needed to wait for the database to be downloaded before adding the model, otherwise it would block the download and never complete.

It kind of works (I click on "iris" and I see the model appear, linked to my cursor). However, when I click to place it this bizarre thing appears:

[image]

tumbili, 2015-10-07:
@v01d Had the same thing happening on my lab pc. On my own pc at home this doesn't happen. Just checking what the problem might be.

v01d, 2015-10-07:
It looks like the rotor_* components are drawn as boxes. I can see two visual elements in the tree, on of type box and another of type mesh. I'm not sure how it should appear.

v01d, 2015-10-07:
I removed the rotor visuals and the boxes disappear. 

By the way, does this require anything on the pixhawk side? I'm doing make sitl_quad and it is not connecting. Should it work over UDP?

devbharat, 2015-10-08:
@LorenzMeier For having it play nice with RotorS, we'd probably need to write wrappers for topics and nodes (to use either ros or gazebo::transport) sort of like we do on the pixhawk. Other change would be to actually generate the sdf from xacro at runtime, while I use a script to generate the model sdf from .xacro separately ahead of time.

thomasgubler, 2015-10-11:
> The IRIS does not fly very well yet, maybe there is a bug somewhere, maybe it's just tuning, we have to see. 

Which mixer is used in this setup? See the difference between the multiplatform iris mixer values and the default iris (quad_w) values.

devbharat, 2015-10-11:
Also, I've always seen it to fly much better with attitude_estimator_ekf than with attitude_estimator_q for some reason. @tumbili  maybe you can try that and bug out why that is so.

devbharat, 2015-10-12:
@tumbili can you confirm that the quad doesn't fly well with attitude_estimator_ekf ? I can fly with it just fine, except for the joystick issues.

tumbili, 2015-10-12:
@devbharat I will try today when Joystick is finally working. @thomasgubler Good point!

tumbili, 2015-10-14:
@devbharat Do you have any idea why yawing does not work. I can see the control signal in QGC but the IRIS does just not yaw. Not sure what the problem is.

devbharat, 2015-10-14:
Can you check the command/motor_speed of the rpms of the motors change when you yaw ? Also, can you compare the mixer you use against the dummy mixer that was used with ROS sitl (in src/platforms/ROS/nodes/mixer

tumbili, 2015-10-16:
@devbharat gazebo is now working for me on master. The only issue I still have is that I cannot control yaw.

devbharat, 2015-10-16:
[image]

Yaw control works for me(Current px4 master, sen_board_rot 8, using attitude_estimator_ekf, master sitl_gazebo). The crazy oscillations seen in the middle are when I made the mistake of changing tabs on qgcs.

LorenzMeier, 2015-10-17:
@devbharat Your state is incompatible with recent Gazebo versions.

devbharat, 2015-10-17:
Ah! Now I get it! I've been using 4.1 all this time.

devbharat, 2015-10-23:
Issues with gazebo6 now fixed on https://github.com/wingtra/sitl_gazebo

closed 2015-11-30 by tumbili