PX4/PX4-Autopilot #2957: PX4 on STM32F4Discovery Pinout

Pedro-Roque, 2015-10-07:
Good morning,

I would like to know what is the pinout for the PX4 Firmware running on the STM32F4Discovery, so I can make a breakout board to use it as Autopilot.

Best,

LorenzMeier, 2015-10-07:
If you intent to work on a development board you're expected to be able to read the board config accordingly. If you just want to get a low-cost solution, I recommend looking at low-cost Pixhawk derivatives: http://www.hobbyking.com/hobbyking/store/__86437__PixFalcon_Micro_PX4_Autopilot_plus_Micro_M8N_GPS_and_Mega_PBD_Power_Module.html

Adding all the breakouts will cost you significantly more than just getting one of the low-cost boards.

closed 2015-10-07 by LorenzMeier

Pedro-Roque, 2015-10-07:
@LorenzMeier Will check that. It should have been the first place to look for! 

Best,