PX4/PX4-Autopilot #2964: FW auto takeoff and landing on runway

AndreasAntener, 2015-10-09:
This should sum up the last TODO/issues that need to be taken care of before we merge runway takeoff and runway landing with terrain estimator.

There is a new branch with the state we flew today: https://github.com/PX4/Firmware/tree/runway_takeoff_before

All in all it worked really well!

TODOs:
- [ ] steer on the ground after landing (and use the navigator with a virtual waypoint ahead, like for takeoff and posctl)
- [ ] parameter for target landing airspeed (factor)

Issues:
- [ ] after landing it throttles up again for a brief moment
- [ ] the switch to mission has to be really fast for the takeoff to work, if assisted is triggered in between it doesnt work

@tumbili please update if I forgot things

AndreasAntener, 2015-10-09:
Log of one of the flights from today: http://logs.uaventure.com/view/fWC9YAEzQLX6z5Yqczwjjj

tumbili, 2015-10-19:
@AndreasAntener @SimonWilks I started fixing some issues, some comments on that:
1) disabling navigation on landing happens too early and is triggered by WP distance instead of height (land_heading_hold_horizontal_distance):

The plane will now only go into pure heading hold mode after we have started flaring. This will make sure that we are not pushed away from the runway by crosswinds.

2) during a landing the fixed wing position controller always sticks to using the terrain estimate if enabled even if the estimate is not valid any more

The position controller will now only use the terrain estimate if it's valid. During a landing it can switch between terrain estimation or normal ground estimation, depending on the validity of the terrain estimate. This is the best thing we can do at the moment. Given that the difference between terrain estimate and normal ground estimate based on altitude should be relatively small this shouldn't be a bit issue.

3) I need a log to find out what is happening with the rudder in stabilized mode, can you send me a link please?

tumbili, 2015-10-19:
@AndreasAntener @SimonWilks Could you give me some more information on this issue:

the switch to mission has to be really fast for the takeoff to work, if assisted is triggered in between it doesn't work

What exactly happens? If you have a log then that would be even better :-)

SimonWilks, 2015-10-19:
You can get the log file here: http://logs.uaventure.com/view/eb266UWc3mWxAaQNbzPchX

Can you send a message back to the ground station if the terrain estimate is no longer valid? It would be good to at least get an audio warning. Alternatively (or additionally) a buzzer error tone could be useful. Basically we want to make the operator aware that something has gone at least partially wrong.

tumbili, 2015-10-19:
@SimonWilks @AndreasAntener Think I found the bug which lead to rudder output in stabilized.
This one was important! If you guys can manage to do a bench test, that would be great!

SimonWilks, 2015-10-19:
@tumbili Cool! I can do a bench test.

tumbili, 2015-10-19:
Not yet, once sec.

tumbili, 2015-10-19:
Ok, good to go.

AndreasAntener, 2015-10-19:
@tumbili here's a log where we have one or two occasions where the main mode switches to ALTCTL before MISSION.
The effect was that it didn't run the takeoff code then, but just started to throttle up.

thomasgubler, 2015-10-19:
> during a landing the fixed wing position controller always sticks to using the terrain estimate if enabled even if the estimate is not valid any more. This has to be rethought because if the estimate is just wrong then the landing won't be nice. Either we switch back to using the normal estimate based on gps and baro or we handle this as failure and do e.g a climbout. See here https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L966 @thomasgubler What do you think?

I see that https @tumbili addressed this here http://github.com/PX4/Firmware/commit/547ef70efe364c3eed5f39c5167bb910b3b27d26

The reason for the old implementation was that the old landing estimator would *sometimes* go invalid again for a short time after becoming valid (in the beginning of the landing phase, depending on the terrain). The estimate never became invalid afterwards for the typical use case. (Of course it's possible, but the whole code here is lacking failure handling anyway). We forced it to stick with the estimate to avoid large jumps in the altitude error.

I guess the safest and most natural thing to do would be a go-around if the terrain estimate becomes invalid. This of course leads to further questions such as: what do you do when the landing fails for n times? Should we force/advise the operator to land it manually? Should we fall back to waypoint altitude landing then? Should this code partly be moved to the navigator and commander (sensor failure handling)? etc.

Another idea would be to let the terrain estimate drift smoothly to the landing waypoint altitude if there are no more valid laser ranger finder inputs by e.g. using the landing waypoint altitude as measurement input with a large variance.

AndreasAntener, 2015-10-19:
@tumbili As long as we don't have the failsafe option I guess switching to the GPOS alt "smoothly" would be a good option? With this I think we can go ahead then and implement failsafe in a next iteration.

On the failsafe: We discussed a few options, one is:
- notify user
- if alt < climbout height, climb out (keep the heading during climbout)
- fly back to previous waypoint

But this would need to be very flexible/configurable. There is no guarantee that a climbout or a turn to go around is even possible.

AndreasAntener, 2015-10-19:
> @tumbili here's a log where we have one or two occasions where the main mode switches to ALTCTL before MISSION. http://logs.uaventure.com/view/jfXbitNoZxe5rCi89Kbyi6
The effect was that it didn't run the takeoff code then, but just started to throttle up.

@tumbili I tried reproducing this in HIL just now, but it doesn't seem to be an issue anymore. Might need to re-check that on the actual setup.

LorenzMeier, 2015-11-06:
@tumbili Can we merge / close?

LorenzMeier, 2015-11-19:
Done! Thanks everybody!

closed 2015-11-19 by LorenzMeier