PX4/PX4-Autopilot #2966: Manual control lost with cmake makster

AndreasAntener, 2015-10-10:
@LorenzMeier, @SimonWilks  from user-list:

We tried to fly “cmake master” today but couldn’t get it going. When we tried to take off it lost control (still on the ground) for about 7 seconds where there is no data in the log:

http://logs.uaventure.com/view/if3FLvipzWXFDa9RD9JkXN
https://gist.github.com/AndreasAntener/86e0e67584e8f4294609 (post/preflight)

This was latest master with runway_takeoff and terrain_estimator, both branches we flew combined on the “old” master successfully afterwards.

LorenzMeier, 2015-10-10:
There is this very weird artifact:
```
Processes: 19 total, 3 running, 16 sleeping
CPU usage: 54.57% tasks, 27.72% sched, 17.71% idle
```

The scheduler should never pull 28% load. Not sure what's going on there. @davids5 Any ideas?

LorenzMeier, 2015-10-10:
It is interesting that GPS stops entirely here:
http://logs.uaventure.com/view/if3FLvipzWXFDa9RD9JkXN#GPS_Uncertainty_PLOT

LorenzMeier, 2015-10-10:
One of the things that probably is also worth addressing is to rate-limit the prediction rate of the EKF to ~50-100 Hz to manage its load:

```
Processes: 20 total, 2 running, 18 sleeping
CPU usage: 88.15% tasks, 0.69% sched, 11.15% idle
Uptime: 1239.725s total, 227.274s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  227273 11.153     0/    0   0 (  0)  READY 
   1 hpwork                      30881  2.461   836/ 1792 192 (192)  w:sem 
   2 lpwork                       5324  0.307   604/ 1792  50 ( 50)  READY 
   3 init                         1581  0.000  1332/ 2496 100 (100)  w:sem 
 175 top                           143  2.000  1228/ 1696 100 (100)  RUN   
  80 dataman                        23  0.000   652/ 1192  90 ( 90)  w:sem 
  92 sensors                     37708  3.076  1652/ 1992 250 (250)  w:sem 
  94 gps                          2220  0.153   708/ 1192 220 (220)  w:sem 
 100 commander                    5127  0.307  2988/ 3392 215 (215)  w:sig 
 103 px4io                       41763  3.461  1028/ 1496 240 (240)  w:sem 
 106 commander_low_prio            718  0.000   748/ 2592  50 ( 50)  w:sem 
 109 mavlink_if0                 11303  0.846  2156/ 2392 100 (100)  w:sig 
 110 mavlink_rcv_if0               348  0.000  1028/ 2096 175 (175)  w:sem 
 122 sdlog2                       3004  0.153  2068/ 2992  70 ( 70)  READY 
 147 ekf_att_pos_estimator      817876 66.307  3428/ 4792 235 (235)  w:sem 
 149 fw_att_control              26375  2.153   876/ 1296 250 (250)  w:sem 
 151 fw_pos_control_l1            5839  0.461   768/ 1296 250 (250)  w:sem 
 155 navigator                   19367  1.538   916/ 1496 105 (105)  w:sem 
 171 mavlink_if1                  2757  4.923  2156/ 2392 100 (100)  READY 
 172 mavlink_rcv_if1                32  0.000  1156/ 2096 175 (175)  w:sem 
```

LorenzMeier, 2015-10-10:
I suspect that  this one here puts this to rest:
https://github.com/PX4/Firmware/pull/2968

tumbili, 2015-10-21:
@LorenzMeier @AndreasAntener Scheduling this for tomorrow

tumbili, 2015-11-06:
@AndreasAntener Flew with master last week, no such issues.

closed 2015-11-06 by tumbili