PX4/PX4-Autopilot #2971: PWM input driver: Excessive ISR handling

LorenzMeier, 2015-10-11:
@AndreasAntener @davids5 It seems that the PWM input capture driver can get into a state where it draws significant CPU time, which shows as high scheduler load (because scheduler in fact is the ISR time).

```
Processes: 19 total, 3 running, 16 sleeping
CPU usage: 54.57% tasks, 27.72% sched, 17.71% idle
```

I suspect the offending code to be here:
https://github.com/PX4/Firmware/blob/master/src/drivers/pwm_input/pwm_input.cpp

LorenzMeier, 2015-10-11:
@davids5 Could you vet the ISR here? I'm wondering if it keeps firing once an error or overrun flag is set which we're not clearing as we should.
https://github.com/PX4/Firmware/blob/master/src/drivers/pwm_input/pwm_input.cpp#L539-L553

LorenzMeier, 2015-10-11:
@AndreasAntener @mhkabir Steps to reproduce would be great.

mhkabir, 2015-10-11:
1. ```param set SENS_EN_LL40LS 1```
2. Make sure the EKF is running (higher CPU load than inav.) I'm personally using James's estimator, which is just slightly higher than inav w.r.t load. Unsure if this is a factor
3. I'm running UAVCAN dynamic node ID server from boot. ``` param set UAVCAN_ENABLE 2``` (again  memory usage factor)
4. If you have a lidar connected, you can try trying to freeze it by covering the lenses, which should trigger the driver reset (and excessive ISR firing if Lorenz is right ^) Or just having nothing connected would do the trick as well if the driver is running.

LorenzMeier, 2015-10-11:
Just running without anything doesn't do it. So its somehow related to the Lidar being connected.

AndreasAntener, 2015-10-12:
I'll see if I can reproduce.

AndreasAntener, 2015-10-12:
So I did a bench test with the same code we used, but I don't know if I was able to trigger it: 
https://gist.github.com/AndreasAntener/d84488fe8fe5850ae965

In the post flight log you see a high scheduler load. But I watched with `top` during the test and never saw the value go over 1 %.

I didn't switch to `mission`, the load stayed at 93-95%. I assume that in our failed attempt the mission processing used up the last few percent when we switched to auto.

LorenzMeier, 2015-11-06:
@davids5 Could you please have a look?

mhkabir, 2015-11-07:
Something of note here : If you run ```pwm_input test``` with no lidar connected, it will freeze the whole system for a few seconds. 

Should  be worthwhile checking out.

davids5, 2015-11-07:
@LorenzMeier will do as soon as I can.

davids5, 2015-11-12:
@LorenzMeier, @mhkabir, @AndreasAntener 

I have had a chance to review this and instrumented it to look for anything wonky.  

With a 1Khz square wave as an input:

[image]

We can see the ISR total time is 3.43 uS (2.86uS is the force to the ring buffer)

[image]

I have run the input at 1, 10 , 100 kHz. 

My Configuration runs at 39-40% with 0 Hz input. 
I have peaks to 41% at 1Kh
I have peaks to 43% at 10Kh
I have peaks to 93% at 100Kh

At 100 kHz the system still runs, but obviously we can not process that number of ISR per second with all the other ISR's running - hence the skips below.

[image]

So I am wondering what the signal from the Lidar looks like? 

Could the issue be caused by noise on the input leaving it chattering?

What other steps can I do to reproduce the load?

davids5, 2015-11-12:
As far as the @mhkabir point out https://github.com/PX4/Firmware/issues/2971#issuecomment-154615122.

The read does not block on an empty queue. So the code runs to completion 

A ::usleep(1000) in the loop of pwmin_test would allow it to play well with others :)

LorenzMeier, 2015-12-02:
I added the usleep in f87407e. Moving this to a later release as we can't reproduce this. @AndreasAntener Please review your flight logs for high scheduler load and explain to us how to repro.

LorenzMeier, 2017-04-10:
This is stable right now but since we anyway wanted to disable this when other inputs are active I'm keeping this for now.

closed 2018-02-22 by PX4BuildBot