PX4/PX4-Autopilot #2972: Bad dependencies for parameters.xml

mcharleb, 2015-10-11:
make posix
Then have a look at build_posix_sitl_simple/CMakeFiles/xml_gen.dir/build.make

parameters.xml depends on EVERY c, cpp and h file under Firmware/src including all example directories, all of Eigen, all platforms. The path passed to the parameter generation script needs to be more specific.

These excess dependencies are breaking a build of qurt following a build for posix.

dagar, 2015-10-11:
Would `*_params.c` catch everything?

DonLakeFlyer, 2015-10-11:
That would seem dangerous to me. Param define macros can be anywhere.

dagar, 2015-10-11:
I should have grepped first. Yes, they're all over the place.
modules/mavlink/mavlink.c, modules/fixedwing_backside/params.c, modules/sdlog2/sdlog2.c, etc

DonLakeFlyer, 2015-10-11:
Is it worth the effort for the time saved to have dependencies on this?

LorenzMeier, 2015-10-11:
well, the trivial fix is to exclude external libraries. Then we'll see what we end up with.

@mcharleb We need to get integration testing up and running for your platform, there is no other way to maintain this. Is there someone on your end with *significant* time available to get this up with me?

LorenzMeier, 2015-10-12:
Fixed by @mcharleb - thanks!

closed 2015-10-12 by LorenzMeier