PX4/PX4-Autopilot #2986: STM32F4Discovery Firmware Pinout for PWM

Pedro-Roque, 2015-10-12:
Hello all,

I am looking for some explanation on PWM I/O for STM32F4Discovery Firmware port. The board.h is here:
https://github.com/PX4/Firmware/blob/master/nuttx-configs/px4-stm32f4discovery/include/board.h

I can't understand if the PWM channels on the firmware support all channels / pins on the hardware.

Best regards, and lots of kudos!

Pedro-Roque, 2015-10-14:
Up @LorenzMeier

LorenzMeier, 2015-10-14:
If you use a dev board you're supposed to know how to read OS / timer configs and schematics. Otherwise use a low-cost autopilot like the Pixfalcon, which will be less expensive than your dev board + sensor breakouts.

closed 2015-10-14 by LorenzMeier

Pedro-Roque, 2015-10-14:
@LorenzMeier , I know how to read those, but what I need to know is if the hardware timers are all available on the PX4 Firmware for this board.

LorenzMeier, 2015-10-14:
src/drivers/px4fmu has all the details on PWM generation.