PX4/PX4-Autopilot #2987: SITL master broken

tumbili, 2015-10-13:
SITL is broken with the latest master, already a while ago.
I have confirmed that this has nothing to do with the noise addition to jMAVSim because I flew successfully with a stable version of the firmware which we used for the ELC workshop
https://github.com/PX4/Firmware/tree/stable_workshop_elc

tumbili, 2015-10-14:
The problem seems to be related to the velocity estimate of the position controller.
If you have a look at the estimated velocities and the gps velocities you can see a huge mismatch.
Attitude estimation seems to be fine. If I remove the bias estimation (no compensation) then the quad flies much better but still the logs show that the estimation does not work properly.

Another thing that I've realized is that the loggs suggests gps data to come in at very irregular time instances and also with huge delays ( > 1s). However if I put a printf in the estimator where position gets updated I get a very regular 5Hz pattern or so.

mswingtra, 2015-10-14:
Hi @tumbili, @devbharat asked me to post these here. He used attitude_estimator_ekf in SITL. Basically for most of the sessions estimated velocities and gps velocities are pretty similar to each other as it should be. 

SITL:
[image]
[image]
[image]

tumbili, 2015-10-14:
@devbharat @mswingtra Thats strange! I tried flying ekf with latest SITL master with latest jMAVSim and the quad just shakes around like crazy.

devbharat, 2015-10-14:
That is what I had reported to you a few days ago, and also on the devcall. These logs (if fact any flying sitl log) are those done before we updated the subtree. 

You are correct about the shakes around like crazy, both jmavsim and gazebo.

jgoppert, 2015-10-14:
I didn't realize we had sitl logging. Is this documented somewhere?  Does
sdlog2 just work?

devbharat, 2015-10-14:
Yes, you can check the command in sitl init rcs. It's the same I think.

tumbili, 2015-10-15:
[image]

tumbili, 2015-10-15:
@LorenzMeier @devbharat This is a log showing the difference between the broken master and working beta. I flew both in jMAVSim in manual, applying full thrust and at some point setting thrust to zero again. So the settings was almost the same in both flights. I have plotted most of the relevant data for velocity estimation in the z axis. All data looks consistent, I can't see anything abnormal. Note however, that the velocity estimate from the log of master is wrong (figure on the left side).

tumbili, 2015-10-15:
@devbharat Fixed by this #3000  Can you do a test and close if the issue is fixed for you?

devbharat, 2015-10-15:
Thanks, will check and close.

tumbili, 2015-10-16:
I checked this using INAV and jMAVSim. Everything went well, POSCTRL, AUTO, ...
However, I tried using the ekf and that gives me a segfault.

closed 2015-11-30 by tumbili