PX4/PX4-Autopilot #2992: SITL: double check logging app

tumbili, 2015-10-14:
@LorenzMeier @mcharleb As mentioned in #2987 there might be a problem with logging.
There is also a problem with argument handling in posix, e.g this gives a segmentation fault:
sdlog2 start -r 100 -e -t -a -b 10

however, this works
sdlog2 start -r 100 -b 10 -e -t -a

A printf shows that with the first method the argument for the option "b" is empty. So it looks like there may be some limit of arguments to pass for posix.

tumbili, 2015-10-14:
I haven't spent too much time on this but also I believe I haven't seen the effect of different logging rates in the logs.

mcharleb, 2015-10-21:
@tumbili Yes, I think there is a limit in the parsed args. Is there a safe upper limit?

LorenzMeier, 2015-10-21:
We need at least 8 args.

LorenzMeier, 2015-11-25:
@mcharleb How many args are you supporting now?

mcharleb, 2015-11-25:
src/platforms/posix/main.cpp:
```
vector<string> appargs(8);
```

LorenzMeier, 2015-11-25:
Bumped it to 10, closing.

closed 2015-11-25 by LorenzMeier