PX4/PX4-Autopilot #2997: Windows master build missing parameters?

dagar, 2015-10-14:
After building and flashing the current master in windows I get this error in QGroundControl.

[image]

dagar, 2015-10-14:
These parameters are present in the generated build_px4fmu-v2_default\src\modules\param\px4_parameters.c and header.

dagar, 2015-10-14:
It looks like a problem with the init scripts.

LorenzMeier, 2015-10-15:
Yes, it requires a boot with the serial console enabled. I suspect that line ending issue on Windows.

dagar, 2015-10-15:
I can't find any line ending differences in all of build_px4fmu-v2_default/tmp/ROMFS/px4fmu_common/init.d/, but rc.autostart is missing.

commit 7d63920f (2015-10-15): srcparser.py split mixed windows path with os.path.split

-fixes #2997
closed 2015-10-15 by dagar