PX4/PX4-Autopilot #2999: hrt call interval too short

tumbili, 2015-10-15:
Trying SITL with latest master give me this warning all the time:
ERROR hrt_call_internal interval too short: 0

LorenzMeier, 2015-10-15:
@mcharleb As a result of this the sensor API stops updating. I think we need to spend some quality time here and address this.

LorenzMeier, 2015-10-15:
As the bare minimum we need to have a facility telling us which call is scheduled with zero.

LorenzMeier, 2015-10-15:
I find this pattern remarkable:
```
WARN  sensors updated
WARN  accel pub
WARN  gyro pub
WARN  accel pub
WARN  gyro pub
WARN  accel pub
WARN  gyro pub
WARN  accel pub
WARN  gyro pub
WARN  sensors updated
WARN  accel pub
WARN  gyro pub
WARN  accel pub
WARN  gyro pub
WARN  accel pub
WARN  gyro pub
WARN  accel pub
WARN  gyro pub
```

We should not read N times the same sample, rather we should back off if there is no new data.

tumbili, 2015-10-15:
@LorenzMeier The reason why this error is only present in the latest master is due to this commit 110a630 So the problem has always been there, it was just never printed.

LorenzMeier, 2015-10-15:
Eventually it moves into this degenerate state where the simulated accel / gyro stop updating, but the simulation task keeps writing new values.

```
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  sensors updated
WARN  WARNING: degenerate accel / mag!
```

LorenzMeier, 2015-10-15:
This is the instrumentation diff:

```diff
diff --git a/src/modules/simulator/simulator_mavlink.cpp b/src/modules/simulator/simulator_mavlink.cpp
index a930d78..597fc89 100644
--- a/src/modules/simulator/simulator_mavlink.cpp
+++ b/src/modules/simulator/simulator_mavlink.cpp
@@ -210,6 +210,8 @@ void Simulator::update_sensors(mavlink_hil_sensor_t *imu)
        airspeed.diff_pressure = imu->diff_pressure;
 
        write_airspeed_data((void *)&airspeed);
+
+       PX4_WARN("sensors updated");
 }
 
 void Simulator::update_gps(mavlink_hil_gps_t *gps_sim)
@@ -240,6 +242,7 @@ void Simulator::handle_message(mavlink_message_t *msg, bool publish)
 
                if (publish) {
                        publish_sensor_topics(&imu);
+                       PX4_WARN("sensors published");
                }
 
                update_sensors(&imu);
diff --git a/src/platforms/posix/drivers/gyrosim/gyrosim.cpp b/src/platforms/posix/drivers/gyrosim/gyrosim.cpp
index cbe1dbe..816b5ca 100644
--- a/src/platforms/posix/drivers/gyrosim/gyrosim.cpp
+++ b/src/platforms/posix/drivers/gyrosim/gyrosim.cpp
@@ -1108,11 +1108,13 @@ GYROSIM::measure()
                perf_begin(_system_latency_perf);
                /* publish it */
                orb_publish(ORB_ID(sensor_accel), _accel_topic, &arb);
+               PX4_WARN("accel pub");
        }
 
        if (!(_pub_blocked)) {
                /* publish it */
                orb_publish(ORB_ID(sensor_gyro), _gyro->_gyro_topic, &grb);
+               PX4_WARN("gyro pub");
        }
 
        /* stop measuring */
```

mcharleb, 2015-10-18:
See PR #3020

This should help to see where the 0 reschedule is coming from

jgoppert, 2015-10-20:
@tumbili, @mcharleb, @LorenzMeier Anyone have time to help debug this? I started it up with gdb, but it is very unresponsive even in gdb, and breaking it just stops at random places in the code, not like it is stuck in a loop.

```bash
(gdb) 
#0  sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
#1  0x000000000042d68b in hrt_work_lock ()
    at /home/jgoppert/git/drone-crew/Firmware/src/platforms/posix/include/hrt_work.h:53
#2  hrt_work_cancel (work=0x2aaaacee2df0)
    at /home/jgoppert/git/drone-crew/Firmware/src/platforms/posix/work_queue/hrt_work_cancel.c:94
#3  0x000000000042a315 in px4_poll (fds=0x2aaaacee2e98, nfds=1, timeout=100)
    at /home/jgoppert/git/drone-crew/Firmware/src/drivers/device/vdev_posix.cpp:246
#4  0x0000000000448814 in Simulator::send (this=0x2aaab0000900)
    at /home/jgoppert/git/drone-crew/Firmware/src/modules/simulator/simulator_mavlink.cpp:353
#5  0x0000000000448879 in Simulator::sending_trampoline ()
    at /home/jgoppert/git/drone-crew/Firmware/src/modules/simulator/simulator_mavlink.cpp:339
#6  0x00002aaaaacd8182 in start_thread (arg=0x2aaaacee3700)
    at pthread_create.c:312
#7  0x00002aaaab7fa47d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) c
Continuing.
^C^C^C^C
Program received signal SIGINT, Interrupt.
[Switching to Thread 0x2aaaad2e5700 (LWP 2615)]
sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
85	in ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S
(gdb) Quit
(gdb) where
#0  sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
#1  0x000000000049929a in task_main (argc=<optimized out>, 
    argv=<optimized out>)
    at /home/jgoppert/git/drone-crew/Firmware/src/modules/dataman/dataman.c:788
#2  0x000000000042cf78 in entry_adapter (ptr=0x718500)
    at /home/jgoppert/git/drone-crew/Firmware/src/platforms/posix/px4_layer/px4_posix_tasks.cpp:87
#3  0x00002aaaaacd8182 in start_thread (arg=0x2aaaad2e5700)
    at pthread_create.c:312
#4  0x00002aaaab7fa47d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) c
Continuing.
^C^C^C^C
Program received signal SIGINT, Interrupt.
[Switching to Thread 0x2aaaacee3700 (LWP 2575)]
sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
85	in ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S
(gdb) Quit
(gdb) where
#0  sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
#1  0x000000000042d68b in hrt_work_lock ()
    at /home/jgoppert/git/drone-crew/Firmware/src/platforms/posix/include/hrt_work.h:53
#2  hrt_work_cancel (work=0x2aaaacee2df0)
    at /home/jgoppert/git/drone-crew/Firmware/src/platforms/posix/work_queue/hrt_work_cancel.c:94
#3  0x000000000042a315 in px4_poll (fds=0x2aaaacee2e98, nfds=1, timeout=100)
    at /home/jgoppert/git/drone-crew/Firmware/src/drivers/device/vdev_posix.cpp:246
#4  0x0000000000448814 in Simulator::send (this=0x2aaab0000900)
    at /home/jgoppert/git/drone-crew/Firmware/src/modules/simulator/simulator_mavlink.cpp:353
#5  0x0000000000448879 in Simulator::sending_trampoline ()
    at /home/jgoppert/git/drone-crew/Firmware/src/modules/simulator/simulator_mavlink.cpp:339
#6  0x00002aaaaacd8182 in start_thread (arg=0x2aaaacee3700)
    at pthread_create.c:312
#7  0x00002aaaab7fa47d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) c
Continuing.
^C^C^C^C
Program received signal SIGINT, Interrupt.
[Switching to Thread 0x2aaaad2e5700 (LWP 2615)]
sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
85	in ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S
(gdb) Quit
(gdb) where
#0  sem_wait () at ../nptl/sysdeps/unix/sysv/linux/x86_64/sem_wait.S:85
#1  0x000000000049929a in task_main (argc=<optimized out>, 
    argv=<optimized out>)
    at /home/jgoppert/git/drone-crew/Firmware/src/modules/dataman/dataman.c:788
#2  0x000000000042cf78 in entry_adapter (ptr=0x718500)
    at /home/jgoppert/git/drone-crew/Firmware/src/platforms/posix/px4_layer/px4_posix_tasks.cpp:87
#3  0x00002aaaaacd8182 in start_thread (arg=0x2aaaad2e5700)
    at pthread_create.c:312
#4  0x00002aaaab7fa47d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) 

```

LorenzMeier, 2015-10-20:
@mcharleb This now shows on Linux as well. I think it would be good if we could start to differentiate instances where the call was made with an interval of zero (so the outermost logic requested it to fire immediately) or if it was scheduled with non-zero but now overran its deadline. E.g. this call is fine to be scheduled with zero:
https://github.com/PX4/Firmware/blob/afb68cb0abee4034582f93d7f383c287b9c9a109/src/drivers/stm32/tone_alarm/tone_alarm.cpp#L590

LorenzMeier, 2015-10-20:
@jgoppert Try running it with this. Part of it is debugging noise.

mcharleb, 2016-04-12:
@LorenzMeier  Is this still an issue?

mcharleb, 2016-04-25:
No longer seen. Closing.

closed 2016-04-25 by mcharleb