PX4/PX4-Autopilot #3001: guard against NaN velocity/position setpoints

v01d, 2015-10-15:
Today I had my quad free-fall in offboard mode and after looking at the sent velocity setpoints, the fall started when my application suddenly sent NaN velocity setpoints. Due to the inherent danger of this condition it might be good to guard against it.

I looked at the code and tried to find whether this is checked somewhere. I found this:
https://github.com/PX4/Firmware/blob/a3552896cae979f1ff7a19731030d68a66844990/src/modules/mavlink/mavlink_receiver.cpp
where "isnan" is checked but only for yaw/yawrate (for some reason). I could provide a PR to add checks for the other setpoints. Should this be done here or in the position controller itself?

I also wonder if, out of extra safety, this check should also be added to mavros.

By the way, what is the proper check to test against this? isnan? PX4_ISFINITE? Also, are infinity values acceptable or should also be guarded?

LorenzMeier, 2015-10-15:
Is this what you want? Maybe requires off board control protection in addition? With some flight testing I'd be happy to merge it:
https://github.com/PX4/Firmware/pull/2814

v01d, 2015-10-15:
Looks like it addresses something similar but I'm a bit lost on why the
addition of remembering the previous setpoint and so on. What is the idea
behind this? Also, do you imagine the same for offboard or should it be
something different?
In my view, I would like to protect against invalid values coming from
outside the board, and I guess the proper reaction by the FCU should be to
safely ignore it. I'm not quite sure how this should work.

LorenzMeier, 2015-11-06:
Closing since fix is merged.

closed 2015-11-06 by LorenzMeier