PX4/PX4-Autopilot #3011: Bring in error state Kalman filter

LorenzMeier, 2015-10-18:


tumbili, 2015-11-06:
## Main features ##
*  Use quaternion error state, after every update the state (3 components) is reset to zero.
Before resetting we add the state to the absolute quaternion.
Doing so we have one attitude state less and we always linearise attitude around the neutral pose.

* The EKF is run on a delayed time horizon (about 0.25s). Therefore, we need to buffer all data.

* Use a complementary filter to predict the real-time states. Do the correction step of this filter based on the measurements and the output state of the EKF at the delayed time horizon. Since the ekf_state is delayed by approx. 0.25s we need to use the buffered measurement data to calculate the innovation.

The reason why we are using the ekf output to perform the correction of the complementary filter is that it will already be very smooth and we can therefore avoid sawtooth effect.
Running the ekf delayed makes sure that we can fuse all measurements at the correct time instant.

## Relevant Links to APM Implementation##

#### Code ####
https://github.com/diydrones/ardupilot/tree/master/libraries/AP_NavEKF2
(everything is inside there)

#### Main filter update function ####
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_core.cpp#L344

#### Complementary filter prediction (real-time) ####
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_core.cpp#L566

#### EKF State prediction (delayed) ####
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_NavEKF2/AP_NavEKF2_core.cpp#L424

## Relevant Links to PX4 Implementation##
#### PX4 Code location ####
https://github.com/LorenzMeier/ecl

tumbili, 2015-11-06:
@yyassa FYI

jgoppert, 2015-11-06:
@tumbili , worth noting that https://github.com/PX4/Firmware/pull/3114 has a similar attitude update, I used the small angle 3 state quaternion error approach, still a WIP though.

tumbili, 2016-02-14:
Done!

closed 2016-02-14 by tumbili