PX4/PX4-Autopilot #3022: "lgamma" function not recognized under GCC 4.9.x

ir-jon, 2015-10-19:
During build of target "make px4fmu-v2_default", build fails at ~60% of dependency scan. Error occurs in <random.h>, which is included in "Firmware/src/modules/ekf_att_pos_estimator/estimator_22states.cpp" via <algorithm.h> on Line 46. Tested on "arm-none-eabi-gcc 4.9.3 20150529 (release)" running under Ubuntu 14.04 LTS. Full error message below:

[image]

v01d, 2015-10-19:
I found the same problem recently, after inadvertently updating my gcc-arm. I had 4.9 2014_q4 (which was working) and with 2015_q2 it was broken. I reverted and everything went back to normal. 
If it is not possible to fix the problem with this compiler, maybe it would be nice to report an error during building that this compiler is not supported.

LorenzMeier, 2015-10-19:
A simple patch would have done 8).

closed 2015-10-19 by LorenzMeier

v01d, 2015-10-19:
Sorry, I'm lost, does that commit fix this?