PX4/PX4-Autopilot #3026: Missing stm32f4discovery make targets

lchish, 2015-10-19:
Hi all,

I'm following these instructions https://pixhawk.org/modules/stm32f4discovery trying to get the px4 firmware running on my stm32f4 discovery board. 

After switching over to cmake the stm32f4 discovery make targets were not ported over and it is now impossible to build using the latest git revision.

The makefile is missing a px4-stm32f4discovery_default target and cmake/configs is missing a nuttx_px4-stm32f4discovery_default.cmake file.

Here is a quick patch I based off the old config_px4-stm32f4discovery_default.mk target. I'm not sure how well it works I have only tried compiling it.

```patch
diff --git a/Makefile b/Makefile
index b896e51..ad2a0bb 100644
--- a/Makefile
+++ b/Makefile
@@ -110,6 +110,9 @@ px4fmu-v1_default:
 px4fmu-v2_default:
 	$(call cmake-build,nuttx_px4fmu-v2_default)
 
+px4-stm32f4discovery_default:
+	$(call cmake-build,nuttx_px4-stm32f4discovery_default)
+
 px4fmu-v2_simple:
 	$(call cmake-build,nuttx_px4fmu-v2_simple)
 
diff --git a/cmake/configs/nuttx_px4-stm32f4discovery_default.cmake b/cmake/configs/nuttx_px4-stm32f4discovery_default.cmake
new file mode 100644
index 0000000..92e40b7
--- /dev/null
+++ b/cmake/configs/nuttx_px4-stm32f4discovery_default.cmake
@@ -0,0 +1,91 @@
+include(nuttx/px4_impl_nuttx)
+
+set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/toolchains/Toolchain-arm-none-eabi.cmake)
+
+set(config_module_list
+	#
+	# Board support modules
+	#
+	drivers/device
+	drivers/stm32
+	drivers/led
+	drivers/boards/px4-stm32f4discovery
+
+	#
+	# System commands
+	#
+	systemcmds/bl_update
+	systemcmds/mixer
+	systemcmds/param
+	systemcmds/perf
+	systemcmds/reboot
+	systemcmds/top
+	systemcmds/tests
+	systemcmds/config
+	systemcmds/nshterm
+	systemcmds/ver
+
+	#
+	# Library modules
+	#
+	modules/systemlib
+	modules/systemlib/mixer
+	modules/controllib
+	modules/uORB
+
+	#
+	# Libraries
+	#
+	lib/mathlib/CMSIS
+	lib/mathlib
+	lib/mathlib/math/filter
+	lib/ecl
+	lib/external_lgpl
+	lib/geo
+	lib/conversion
+	platforms/nuttx
+
+	# had to add for cmake, not sure why wasn't in original config
+	platforms/common
+	platforms/nuttx/px4_layer
+
+	#
+	# Demo apps
+	#
+	#examples/math_demo
+	# Tutorial code from
+	# https://px4.io/dev/px4_simple_app
+	examples/px4_simple_app
+
+	# Tutorial code from
+	# https://px4.io/dev/daemon
+	#examples/px4_daemon_app
+
+	# Tutorial code from
+	# https://px4.io/dev/debug_values
+	#examples/px4_mavlink_debug
+
+	# Tutorial code from
+	# https://px4.io/dev/example_fixedwing_control
+	#examples/fixedwing_control
+
+	# Hardware test
+	#examples/hwtest
+)
+
+set(config_extra_builtin_cmds
+	serdis
+	sercon
+	)
+
+set(config_io_extra_libs
+	#${CMAKE_SOURCE_DIR}/src/lib/mathlib/CMSIS/libarm_cortexM3l_math.a
+	)
+
+add_custom_target(sercon)
+set_target_properties(sercon PROPERTIES
+	MAIN "sercon" STACK "2048")
+
+add_custom_target(serdis)
+set_target_properties(serdis PROPERTIES
+	MAIN "serdis" STACK "2048")
```

LorenzMeier, 2015-10-21:
If you send a pull request we're happy to merge it.

lchish, 2015-10-21:
At the moment I'm unable to get it to configure unless I add config_io_board to the cmake file. The nuttx cmake files require a src/modules/px4iofirmware/{config_io_board}_default.bin to exist:

```
make -C /run/media/leslie/arch/Firmware/build_px4-stm32f4discovery_default -j4 --no-print-directory 
make[3]: *** No rule to make target 'src/modules/px4iofirmware/_default.bin', needed by 'src/firmware/nuttx/extras/_default.bin'.  Stop.
```

It might be worth creating a new issue to allow nuttx to build without requiring px4iofirmware and then take my pull request to add the stm32discovery make targets.

lchish, 2015-10-21:
I've pushed up some more changes to remove the px4iofirmware requirement and managed to get the build to run and upload onto my discovery board.

LorenzMeier, 2016-02-18:
Thanks! Now in the tree for a while already.

closed 2016-02-18 by LorenzMeier