PX4/PX4-Autopilot #3036: UAVCAN module in Python

LorenzMeier, 2015-10-21:
@pavel-kirienko I get this error with Python 2.7.9 - any idea what could be wrong here?

```sh
[  2%] Built target generate_topic_listener
Scanning dependencies of target git_eigen
[  2%] Generating git_init_src_lib_eigen.stamp
Running from the source directory
Traceback (most recent call last):
  File "dsdl_compiler/libuavcan_dsdlc", line 59, in <module>
    from libuavcan_dsdl_compiler import run as dsdlc_run
  File "/home/drone/src/px4test/76963e82ab032030042fa385cda331284357c85e/Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py", line 17, in <module>
    from uavcan import dsdl
ImportError: No module named uavcan
src/modules/uavcan/libuavcan/libuavcan/CMakeFiles/libuavcan_dsdlc.dir/build.make:49: recipe for target 'src/modules/uavcan/libuavcan/libuavcan/CMakeFiles/libuavcan_dsdlc' failed
make[3]: *** [src/modules/uavcan/libuavcan/libuavcan/CMakeFiles/libuavcan_dsdlc] Error 1
CMakeFiles/Makefile2:3223: recipe for target 'src/modules/uavcan/libuavcan/libuavcan/CMakeFiles/libuavcan_dsdlc.dir/all' failed
make[2]: *** [src/modules/uavcan/libuavcan/libuavcan/CMakeFiles/libuavcan_dsdlc.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  2%] Built target git_mavlink
[  3%] Built target git_uavcan
[  3%] Built target git_eigen
Makefile:147: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:111: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2
drone@zurich01:~/src/px4test/76963e82ab032030042fa385cda331284357c85e/Firmware$ less /home/drone/src/px4test/76963e82ab032030042fa385cda331284357c85e/Firmware/src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/libuavcan_dsdl_compiler/__init__.py
drone@zurich01:~/src/px4test/76963e82ab032030042fa385cda331284357c85e/Firmware$ python --version
Python 2.7.9
```

pavel-kirienko, 2015-10-21:
Are you sure the git submodules were checked out with `--recursive`? This issue it often caused by mussing pyuavcan submodule.

LorenzMeier, 2015-10-21:
Yes, checked, will re-check

LorenzMeier, 2015-11-06:
Fixed.

closed 2015-11-06 by LorenzMeier