PX4/PX4-Autopilot #3046: Code stability in air on master

LorenzMeier, 2015-10-23:
@tumbili Unfortunately it looks like we have a stability issue which is isolated to fixed wing aircraft on master.

This is a log ending mid-air:
http://logs.uaventure.com/view/KmWNFWsiqxXMHYTR23So46

I suspect the stack size of the EKF estimator. I have adjusted it downward recently after intensely reviewing in-flight stack usage. But its possible there is a code path that I missed during that review. We need to come up with an approach to reproduce this on the bench.

LorenzMeier, 2015-10-23:
We need to address this issue ASAP to make test-flights safer again: https://github.com/PX4/Firmware/issues/3003

LorenzMeier, 2015-10-23:
I've pushed a hotfix in 7fd6d2d. That brings the stack back to the same level as on stable, and most likely fixes the issue.

LorenzMeier, 2015-10-23:
In a crash log on stable the user had connected what looks like USB before he flew. We need to think about the implications of that.

tumbili, 2015-10-23:
Some facts about the crash flight:
- I had the autopilot connected over usb before the flight (did not use telemetry)
- I did not check

tumbili, 2015-10-23:
@LorenzMeier Notice also that the ekf is not working at all on SITL for fw whereas the other estimators work fine with the same setup.

AndreasAntener, 2015-11-01:
Just flown master: http://logs.uaventure.com/view/bzijw7axp3j9Tpqk4sGtXb
Had no problems. My goal actually was to reproduce #2893 but the problem didn't occur.

tumbili, 2015-11-06:
Closing, as test flights went without any problems.

closed 2015-11-06 by tumbili