PX4/PX4-Autopilot #3049: SITL floating point exception

tumbili, 2015-10-23:
@mcharleb @LorenzMeier  We have the problem that floating point exceptions terminate the flight code in SITL.
The floating point exception is happening in the ekf_att_pos_estimator module.
I prepared this branch here for you to reproduce the problem:
https://github.com/PX4/Firmware/tree/floating_point_exception

You just have to run "make run_sitl_quad"

You will probably see the console being spammed with floating point exception messages. This comes from a signal handler for floating point exceptions in the posix mainapp which we have added (see the last commit of the upper branch).

LorenzMeier, 2015-10-23:
Mark, goal is to add backtrace - in case you beat me to it.

LorenzMeier, 2015-11-06:
Fixed.

closed 2015-11-06 by LorenzMeier

yyassa, 2016-04-06:
@LorenzMeier Could you please link to how this was fixed?

I am getting floating point exceptions in SITL  (running an old version of ekf_att_pos_estimator). Have a feeling it might be this.