PX4/PX4-Autopilot #3051: Mission transfer in SITL not working

tumbili, 2015-10-23:
Lots of these messages:
Critical: WPM: REJ. CMD: Busy

Looks like there is a problem with the waypoint manager.

This problem does not exist on the physical hardware (e.g. Pixhawk).

tumbili, 2015-10-26:
@LorenzMeier This is the output of mavlink:
```
WARN  WPM: MISSION_COUNT 2 from ID 255, changing state to MAVLINK_WPM_STATE_GETLIST
WARN  WPM: Send MISSION_REQUEST seq 0 to ID 255
WARN  WPM: MISSION_COUNT 2 from ID 255 (again)
WARN  WPM: Send MISSION_REQUEST seq 0 to ID 255
WARN  WPM: MISSION_ITEM seq 0 received
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: MISSION_ITEM ERROR: seq 0 was not the expected 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Last operation (state=2) timed out, changing state to MAVLINK_WPM_STATE_IDLE

```

tumbili, 2015-10-26:
And on the QGC side you always get this error:
Error during mission communication with Vehicle: Vehicle sent unexpected MISSION_ACK message, error: Unspecified error (MAV_MISSION_ERROR)

tumbili, 2015-10-29:
SITL:
Unknown message from system: 1 message: 245
Unknown message from system: 1 message: 87
Unknown message from system: 1 message: 242
Unknown message from system: 1 message: 44
Unknown message from system: 1 message: 76
Unknown message from system: 1 message: 47
Unknown message from system: 1 message: 36

Pixhawk:
Unknown message from system: 1 message: 36
Unknown message from system: 1 message: 87
QGCMessageBox (unit testing) "Setup" "One or more vehicle components require setup prior to flight."
"BAT_V_EMPTY" QVariant(float, 3.5) 3
"BAT_V_CHARGED" QVariant(float, 4.2) 3
"COM_RC_LOSS_T" QVariant(float, 0.5) 3
"RTL_LAND_DELAY" QVariant(float, 10) 3
"RTL_DESCEND_ALT" QVariant(float, 25) 3
"RTL_RETURN_ALT" QVariant(float, 30) 3
Unknown message from system: 1 message: 76
Unknown message from system: 1 message: 44

tumbili, 2015-10-29:
Here we have printed a timestamp when a mission count message was received: (time ...)

INFO  ToneAlarm::start_note 11375
time: 10072897f
WARN  WPM: MISSION_COUNT 2 from ID 255, changing state to MAVLINK_WPM_STATE_GETLIST
WARN  WPM: Send MISSION_REQUEST seq 0 to ID 255
time: 10072946f
WARN  WPM: MISSION_COUNT 2 from ID 255 (again)
WARN  WPM: Send MISSION_REQUEST seq 0 to ID 255
WARN  WPM: MISSION_ITEM seq 0 received
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: MISSION_ITEM ERROR: seq 0 was not the expected 1
time: 10182850f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
time: 10182868f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
time: 10273734f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
time: 10273762f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
INFO  TONE_SET_ALARM 0
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
time: 10826026f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
time: 10826060f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
time: 11328112f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
time: 11328144f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
time: 11830264f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
time: 11830297f
WARN  WPM: MISSION_COUNT ERROR: busy, already receiving seq 1
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Send MISSION_REQUEST seq 1 to ID 255
WARN  WPM: Last operation (state=2) timed out, changing state to MAVLINK_WPM_STATE_IDLE

DonLakeFlyer, 2015-10-29:
We'll need to log from the QGC side as well when this happens: --logging:MissionManager. Possible UDP implementation problem. Hard to say. Looks like QGC is timing out and retrying.

devbharat, 2015-10-29:
I just ran a hands free (joystick free essentially, that is still painful to me) on a fresh new PC, with latest px4 firmware and latest qgc, and iris did a complete mission. It was beautiful.

devbharat, 2015-10-29:
Though it is not perfect, I can't reproduce it.

tumbili, 2015-10-30:
Possible fixed on the QGC side, see https://github.com/mavlink/qgroundcontrol/pull/2161

tumbili, 2015-10-30:
Fixed, closing.

closed 2015-10-30 by tumbili