PX4/PX4-Autopilot #3066: Make LPE Default Position Estimator

jgoppert, 2015-10-24:


LorenzMeier, 2015-10-24:
@jgoppert It did not pause, but instead it overshot, which is against the design of the system. How fast is LPE publishing? I can see the sawtooth in the estimate clearly.

jgoppert, 2015-10-24:
Update rate is controlled by sensor combined topic, so around 100-200 Hz. The saw tooth I think is from the GPS and inertial positions fighting. If we increase the accel noise, the integrator would give up and start trusting GPS more.

mhkabir, 2015-10-25:
@LorenzMeier @jgoppert With respect to the sawtoothing, I was thinking about the low-pass filter introduced on the controllers to smooth the estimate. Shouldn't any sort of smoothing be done on the estimator side, since it would benefit *all* downstream consumers of the position estimate?

scott-eddy, 2015-10-26:
I would think that running the output of a Kalman filter through a low pass filter is really not a proper solution here.  Ideally we should be able to adjust the tuning knobs of the KF (noise variances, adding states such as accel bias, etc.) in order to get better estimator performance. Filtering the output of one filter with another makes overall tuning more difficult and is not really addressing the problems inherit in the first filter.  

Just my 2-cents

jgoppert, 2015-10-26:
@mhkabir , @scott-eddy  I agree that we have to address the saw-toothing. This was more of an edge case. I flew about 5 times with that code and only saw this once. As @scott-eddy suggested, if we increase the accel noise, it will converge to the GPS trajectory and you won't see any more saw toothing, so this would likely be better than a low pass filter. When a kalman filter saw tooths, two things are fighting, in this case GPS and accel and we don't let either be dominant.

jgoppert, 2015-10-27:
@mhkabir, @LorenzMeier, @scott-eddy 

After the latest fix to the bias estimation I'm now getting excellent results in several live flight tests:

[image]

[image]

[image]

The next item on the list to address should be GPS delays. Here is a minimization of the error between the estimated and GPS signal. The minimum represents the best fit.

[image]

The GPS xy position are lagging about 150 milliseconds while the altitude and velocity components are lagging by about 250 milliseconds.

Flight log: http://logs.uaventure.com/view/L373uGfDQH5iuSA9XiQgGS

LorenzMeier, 2015-10-27:
Hi James,

Awesome! That looks much more like it.

The interpretation of that data should be that the GPS is always lagging 250 ms but the baro is already lagging 100 ms behind the accel, and needs similar compensation. The accel lag itself is about 30 ms with the low passed values you're currently using and more in the 10 ms range with the delta velocities I added recently.

LorenzMeier, 2015-10-27:
Looking at this I also realised that we should be publishing navigation_capabilities from the controller and announcing our min switch radius, which would be current velocity * brake time, where brake time would be max deceleration integrated to hit a value of zero. E.g. for the plot above setting the brake time to 1s would have resulted in 2-5m waypoint switch radius.
https://github.com/PX4/Firmware/blob/master/msg/navigation_capabilities.msg

Set turn distance to current velocity * 1s. The navigator picks max(wp_switch_dist, navigator_switch_dist). That would have led to a smoother trajectory here, and is an easy and quick fix.

scott-eddy, 2015-10-27:
This is looking much better.  Any chance of it flying with optical flow indoors any time soon?  It seems that GPS denied estimation has *understandably* been put on the back burner, but ideally won't be forgotten.  Hopefully with the bias fix we would see improvement in the indoor case as well.

Can anyone work on this?  I am getting a quad with a flow up and running now but it won't be flying for a month probably.

mhkabir, 2015-10-28:
@scott-eddy Working on it. I just came back from vacation. I'll fly and upload a video of pure-flow flight with LPE soon. I have some fixes here on my side which will need to be sent in for things to start working properly again (#3067) :)

LorenzMeier, 2015-11-06:
@jgoppert What is holding this off is the question of the influence of GPS and baro delays on the gyro offset estimate. I would be thrilled to switch to your attitude EKF as well, but I want to first be sure that we fix the root cause with the right solution.

Its clear that a 200 ms phase delay has an influence on the gyro offset estimation, and as long that's not addressed in the position estimator and validated I don't want to start messing with the attitude estimate.

LorenzMeier, 2016-01-06:
Closing this due to lack of activity.

jgoppert, 2016-02-01:
Reopening this to start work to wrap it up.

Goals are now:
1. Analyze GPS delay effect on attitude gyro bias estimation vs. INAV.
2. Determine accuracy of accel bias estimation using fixed offsets in simulation.
3. Test indoor flight with flow.

@LorenzMeier  Please feel free to add to this list.

@TSC21, @BMP-TECH, @mhkabir  If you guys can help me address these let me know. I'd like to finish this quickly.

mhkabir, 2016-02-02:
@jgoppert @LorenzMeier
#3209 has a load of changes implementing number 3 and fixing miscellaneous issues. It would be ideal

jgoppert, 2016-02-02:
@mhkabir Sounds good. I think we need to get some plots of performance vs. INAV as a base line for the flow that it is robust enough for main stream.

jgoppert, 2016-03-17:
@LorenzMeier let me know what else is holding this back.

LorenzMeier, 2016-07-29:
Closing in favour of #4483.

closed 2016-07-29 by LorenzMeier