PX4/PX4-Autopilot #3067: LPE Enhancements and Fixes

mhkabir, 2015-10-25:
LPE has several things we can improve on, opening this to keep track of the TODOs.

[ ] Fix multi-subscription and properly handle rangefinders e.g when the flow board is connected. Range-finders do not work properly in LPE at the moment when more than one is connected (#2965)
[ ] Don't use flight controller gyro for flow body-rotation compensation, doesn't work well. (https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp#L870)
[ ] Add estimation of gyro bias for the flow board's gyro and use it instead for the compensation.
[ ] Use Roman's terrain estimator and differentiate between AGL altitude vs terrain distance.
[ ] Use the terrain estimate for flow scaling, not local Z directly (https://github.com/PX4/Firmware/blob/master/src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp#L875)

@jgoppert Please feel free to edit/add to the list as required. I will be sending in PRs after flight testing for the above.

mhkabir, 2015-10-28:
@jgoppert @LorenzMeier 
I have done some work on this, and here is the working branch : https://github.com/PX4/Firmware/compare/master...mhkabir:lpe_flow

Your comments are welcome. This is still heavily WIP, so I haven't yet opened any pull requests.

mhkabir, 2015-10-28:
@jgoppert With respect to terrain estimation, do we want to use ```terrain_estimation``` from the ECL (by Roman) or just roll our own support inside LPE, with terrain altitude as a filter state?
I'd like to start work on this so that I can get flow working fully.

scott-eddy, 2015-11-06:
@mhkabir I'd like to help test your lpe_flow branch.  To keep things consistent what Flow firmware are you using?  Are you building it from source or installing from QGC?  

Since I'd like to get flying indoors soon I'm going to be characterizing which FMU/Flow setup is currently the best.  @LorenzMeier I'd like to compare this to recently merged #3040.  The user group says that QGC should pull the right binary for the flow, does it matter if it is the daily build of QGC or does stable work?

mhkabir, 2015-11-07:
@scott-eddy 
It'd be great to have your help in testing. I've tried both flow firmwares in fact. The KLT firmware gives a much lower quality metric and we are still yet to decide whether the use of the 'quality' field is even good now. 

With the KLT fw and my lpe_flow branch, it is unlikely that you'll get good performance. LPE uses a quality threshold of 100 to reject bad flow data. With KLT fw in good conditions: 
[image]

I am currently working out whether eliminating quality completely is best.
There are a number of reasons for this :
1. With KLT, the calculation is different from the patch-based method previously employed. It never goes to max quality (255) even under good conditions.
2. Using external rangefinders (e.g Lidar-Lite) gives a performance boost, but the quality field takes into account the sonar, which obviously limits performance, if it is considered a metric.

Further, I'm not quite sure how the changes to the exposure handling on the new Flow firmware works. There is supposed to be 4x improvement, which should improve low-light performance, but maybe it has negatively affected outdoor performance? Cannot test this as QGC no longer has the Video Downlink widget to check the sensor image.

The binary you install from QGC will be the KLT firmware. For the older one, you can flash PX4/Flow master.

mhkabir, 2015-11-08:
@LorenzMeier I was thinking about this : Can we get over the algorithm performance data over to the flight controller? As in feature-count or similar. I think this would be good long-term as well, even though modifications to the i2c frame would be required.
We could/should do this quality calculation on the flight controller side. Or maybe even a proper covariance estimate from the available data. 

I think we need to get this settled soon via a call or chat.

@scott-eddy Any other ideas?

scott-eddy, 2015-11-09:
I actually am of the mindset that we don't necessarily need to make the quality calculation on the flight controller side, we just need to have a better understanding about how the quality metric is being calculated.  For instance a feature count or even a proper co variance estimate is merely a better understood/defined quality metric.  

Given that @LorenzMeier has pointed out before in #1722 and Flow #42 that changing the i2C frame needs to be done with minimal impact to current users I'm guessing altering it to do the quality calculation on the FCU is not going to be merged in.

Is the source for the KLT binary the flow/klt_flow or flow/klt_flow2 on github, or neither of these?  I'm guessing if we understand the quality calculation there we can determine:
a) if the metric is good as it is currently calculated
b) if a threshold of 100 makes sense

mhkabir, 2015-11-10:
I agree with you mostly, but we also need to consider that the flight controller *always has more state information* than the PX4Flow. The quality calculation, if done on the flow side *is* affected by e.g
1. biased gyro readings
2. limited sonar range
A better understanding of the exact calculation would be the best to know though. Making it independent (or if it already is) of the above two would make it just fine, IMO. That said, a properly defined variance metric would be loved.

For current flight-testing purposes I reduced the threshold to 50 on my branch, otherwise too much data was ignored.

scott-eddy, 2015-11-10:
It looks like on the master branch quailty is based on the pixel flow and the gyros only, not the sonar as seen [here](readings: https://github.com/PX4/Flow/blob/3ddba48fddcd4140499b54cccba25994669ac190/src/modules/flow/flow.c#L399).  Also, on the DevCall yesterday Lorenz made a point about how quality really is a non-physical unit less metric that we really shouldn't be hung up about.  While this really indicates it should be re-thought to indeed mean something as a first pass I was also going to make the threshold a tuning parameter. 

Also, I'd appreciate your thoughts on my physical configuration really quick before I begin some testing.  My setup has the flow downward facing on a gimbal on which I'm going to mount a separate CV camera.  I don't think this will be a problem as the gyro compensation is based on the on-board flow gyros.  The only discrepancy I can think of is that the lidar is mounted with a vertical displacement.  I was also going to add this vertical offset as a parameter as well because I see many people having to mount their lidar's below the pixhawk due to physical limitations.  Thoughts on the setup would be appreciated.

jgoppert, 2016-03-17:
I think we handled this with latest lpe changes.

closed 2016-03-17 by jgoppert