PX4/PX4-Autopilot #3080: Implement motor slew rate limiting in mixer

LorenzMeier, 2015-10-27:
@tumbili I've heard now about multiple platforms having issues with extreme slew rates in the actuators. We should implement a per-motor slew rate limiter.

tumbili, 2016-08-04:
@LorenzMeier @jgoppert @kd0aij I'm intending to start with this now. I've also followed the discussion here https://github.com/PX4/Firmware/issues/5203 and I think timing is good to tackle both issues at once.
Here's the plan:
1) Mixer slew rate limiter: Implement simple per motor  slew rate limiting based on max derivative of motor signal. Max slew rate can be adjusted via parameter.

2) Implement static thrust to pwm mapping: Based on the thrust/pwm data from APM we can implement the mapping from desired thrust / torque to actual pwm. We are going to neglect motor dynamics as they are well above the bandwidth of the rate controller. We are also going to neglect the aerodynamic derivatives of the propeller (e.g. influence of angular speed on thrust).  We will have a parameter to influence the actual shape of the thrust to pwm mapping.

LorenzMeier, 2016-08-04:
Sounds great, looking forward to the PR!

tumbili, 2016-10-21:
Implemented in master

closed 2016-10-21 by tumbili

kd0aij, 2016-10-21:
Still no feedback at all here though: http://discuss.px4.io/t/fpv-racer-tuning/969/2