PX4/PX4-Autopilot #3082: Line calculation for POSCTL, convergence north/south

AndreasAntener, 2015-10-27:
@tumbili as discussed

I've observed this previously but didn't follow up on it (https://github.com/PX4/Firmware/issues/2299). When testing variations on runway takeoff we also used the same calculation to draw a line with current heading. It resulted in a wrong actual heading during takeoff:

[image]

The blue lines were calculated by the posctl method, the red line is were it actually should be. Note that both lines match for 90 degrees.

Example calculations (posctl method):
```
Origin: 47.389492,8.0317354

160:
Point behind:
47.389492 - cos(160degrees) * 100/1e6 = 47.3895859693
8.0317354 - sin(160degrees) * 100/1e6 = 8.03170119799

Point ahead:
47.389492 + cos(160degrees) * 3100/1e6 = 47.3865789529
8.0317354 + sin(160degrees) * 3100/1e6 = 8.03279566244

90:
Point behind:
47.389492 - cos(90degrees) * 100/1e6 = 47.389492
8.0317354 - sin(90degrees) * 100/1e6 = 8.0316354

Point ahead:
47.389492 + cos(90degrees) * 3100/1e6 = 47.389492
8.0317354 + sin(90degrees) * 3100/1e6 = 8.0348354
```

Solution:
The calculation we actually want can be found here (great circle distance): http://www.movable-type.co.uk/scripts/latlong.html (part "Destination point given distance and bearing from start point")

- calculate the 2 waypoints with bearing and constant distance (a few hundred m)
- before we reach it, calculate new actual bearing at the end of the line
- use the new bearing for the next pair of waypoints

tumbili, 2015-10-27:
@AndreasAntener Thanks, I'll try to get this in for tomorrow.

tumbili, 2015-10-29:
This functionality is addressed in a commit in #2969 Once we have it working there we can also use the functions for POSCTRL.

tumbili, 2016-02-14:
Fixed!

closed 2016-02-14 by tumbili