PX4/PX4-Autopilot #3084: Gazebo simulator: No MAVLink data received / sent, no output

LorenzMeier, 2015-10-27:
I put this instrumentation into the plugin:

```diff
diff --git a/src/gazebo_mavlink_interface.cpp b/src/gazebo_mavlink_interface.cpp
index 4737ed0..eedfbd8 100644
--- a/src/gazebo_mavlink_interface.cpp
+++ b/src/gazebo_mavlink_interface.cpp
@@ -79,12 +79,16 @@ void GazeboMavlinkInterface::Load(physics::ModelPtr _model, sdf::ElementPtr _sdf
   // udp socket data
   const int _port = UDP_PORT;
 
+  printf("GAZEBO: SOCKET TRY");
+
   // try to setup udp socket for communcation with simulator
   if ((_fd = socket(AF_INET, SOCK_DGRAM, 0)) < 0) {
     printf("create socket failed\n");
     return;
   }
 
+  printf("GAZEBO: SOCKET CREATED");
+
   _srcaddr.sin_family = AF_INET;
   _srcaddr.sin_addr.s_addr = htonl(INADDR_ANY);
   _srcaddr.sin_port = htons(UDP_PORT);
@@ -305,6 +309,7 @@ void GazeboMavlinkInterface::ImuCallback(ImuPtr& imu_message) {
 
 void GazeboMavlinkInterface::pollForMAVLinkMessages()
 {
+  printf("GAZEBO: POLL");
   int len;
   ::poll(&fds[0], (sizeof(fds[0])/sizeof(fds[0])), 0);
   if (fds[0].revents & POLLIN) {
```

It looks like the plugin is not loaded. This is how the model tree looks like for me:
[image]

LorenzMeier, 2015-10-27:
@tumbili Could you share a screenshot of your model and see if you can see the instrumentation on the console? All I get is this:

```
[ 74%] Built target mav_msgs
[ 78%] Built target hello_world
[ 82%] Built target rotors_gazebo_mavlink_interface
[ 87%] Built target rotors_gazebo_controller_interface
[ 91%] Built target rotors_gazebo_imu_plugin
[ 95%] Built target rotors_gazebo_motor_model
[100%] Built target rotors_gazebo_multirotor_base_plugin
[init] shell id: 140735163671296
[init] task name: mainapp

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

Ready to fly.

pxh> INFO  Not using /dev/ttyACM0 for radio control input. Assuming joystick input via MAVLink.
INFO  Waiting for initial data on UDP. Please start the flight simulator to proceed..
```

LorenzMeier, 2015-10-27:
I did dig through this further: What I noticed is that if I set the plugin name to a non-existing plugin it will not abort the boot just silently fail.

devbharat, 2015-10-29:
Maybe verifying that the plugin is loaded through a print in the 'load' function would be a good idea in general. I had also observed that in case a plugin's .so isn't found, it is simply neglected and the rest of the model is loaded (which is why it's better to have those prints on the load functions). Though I have not observed the plugin's suddenly not loading for apparently no reason. What does happen sometimes is when i abruptly shut down udp communication on one end and try to reconnect it, and the previous instance did not release the port.

LorenzMeier, 2015-11-25:
Fixed.

closed 2015-11-25 by LorenzMeier