PX4/PX4-Autopilot #3087: EKF - GPS corner case

mswingtra, 2015-10-28:
On Monday we were doing flight tests and we hit a corner case with GPS fusing in EKF. 
Basically, we took off with a really bad GPS signal (epv above 6) so it was not being fused. We flew a bit in attitude, then switched to altitude. In altitude, the GPS signal finally dropped below 5, so it was good enough to be fused. As last good gps signal was way longer ago than the set threshold of 5 sec, the position and velocity states have been reset (ekf_att_pos_estimator_main.cpp L:1463). And that caused the LPOS_Z to go really bad, which increased error in mc_pos_controller. Thus controller responded with full throttle.
From our side, mistake was not to check gps signal before taking off, but generally it could happen not only to us and it can be quite dangerous. 
If you want to have a look in logging data - http://logs.uaventure.com/view/ktNtuubSU9Lz4oQyv3rGjG

Now, @LorenzMeier told us today that there is now included filtering of velocities in mc_pos_controller that filters out sudden jumps in inputs so that would probably prevent the excessively aggressive response of the controller. However, it does not solve the above case.

LorenzMeier, 2015-10-28:
Thanks for the report. It should have reset the position setpoint when it did update its position. Something we need to look at closer.

LorenzMeier, 2015-11-06:
@mswingtra Could we get your patch?

tumbili, 2015-12-02:
EKF notify about origin reset so that position controller can deal with it.

LorenzMeier, 2016-01-24:
Closing this as won't fix since we move to EKF2.

closed 2016-01-24 by LorenzMeier