PX4/PX4-Autopilot #3094: Verify multicopter position control performance

LorenzMeier, 2015-10-29:
@tumbili This is to track the symptoms you've seen.

tumbili, 2015-10-29:
Postion control performance in the GLOBAL y axis is bad, x axis is very good.

LorenzMeier, 2015-10-29:
One particular observation is that some of the local -> global transformations are based on yaw setpoint, not current yaw. So the position control performance can depend on yaw tracking, which is not good.

LorenzMeier, 2015-10-29:
The map projection / reprojection math should be checked. Did you fix any of that for fixed wing?

tumbili, 2015-10-29:
No, I just introduced new functions, didn't change any of the existing ones.

LorenzMeier, 2015-10-29:
That was not the question. The question is if the existing functions suffer from a similar deficiency.

tumbili, 2015-10-29:
[image]

tumbili, 2015-10-29:
Could also be an estimation problem. This plot shows the derivative of the position (x,y) vs. the velocities. The errors in the y axis are much larger than the ones in the x axis.

tumbili, 2015-10-29:
Ok, if you compare gps velocity with estimated velocity on both axes then the problem is obvious the estimation (possibly caused by false sensor data).

tumbili, 2015-10-29:
Got it, the calculation of the lat, lon values in sitl_gazebo is not correct. It works for once direction because the longitude value of Zurich is relatively small.

tumbili, 2015-10-30:
Solved, closing

closed 2015-10-30 by tumbili