PX4/PX4-Autopilot #3097: Publish all distance sensors via MAVLink

AndreasAntener, 2015-10-29:
@mhkabir @LorenzMeier 
Since all the distance sensors are now publishing thanks to Kabir, #3085, I guess updating the `DISTANCE_SENSOR` message for MAVLink is now the next step. It should subscribe to every instance and publish all with a different ID. Receivers would then need to filter according to type and/or ID.

Will the instance IDs actually stay the same over boots?

For reference, related issues: https://github.com/PX4/Firmware/issues/2480, https://github.com/PX4/Firmware/issues/2297

dagar, 2017-04-28:
Does this need additional mavlink DISTANCE_SENSOR messages or is publishing with different ids acceptable?

mhkabir, 2017-04-28:
We already do this.

closed 2017-04-28 by mhkabir