PX4/PX4-Autopilot #3099: CMake Error at Firmware/CMakeLists.txt

zshn25, 2015-10-30:
I followed the instructions from https://pixhawk.org/dev/ros/sitl

I created and configured the workspace manually. Everything went well untill the last step.

When I run catkin_make, it gave the following error messages

```
CMake Error at Firmware/CMakeLists.txt:158 (include):
  include could not find load file:

    configs/nuttx_px4fmu-v2_default

CMake Error at Firmware/CMakeLists.txt:170 (message):
  configs/nuttx_px4fmu-v2_default must implement px4_os_prebuild_targets

-- Configuring incomplete, errors occurred!
See also "/home/zee/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/zee/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

killerlayup, 2015-11-02:
I have the same issue, pls help me

devbharat, 2015-11-02:
Can you try the updated documentation at dev.px4.io ? Do you specifically want to compile the firmware for ros, or just want a general sitl setup(provided in the updated doc on the link above) ?

zshn25, 2015-11-02:
I want to compile the Firmware for ROS which includes SITL too.

killerlayup, 2015-11-02:
I want to compile the Firmware for ROS which includes SITL too.

devbharat, 2015-11-02:
I am not sure, but I guess the ROS compilation has some issues right now, you could go to previous commit master before the CMake branch was merged in.

blackcoder, 2015-11-02:
Same issue here, anyone knows what changes introduced this recently ?

killerlayup, 2015-11-02:
Same issue here, anyone knows what changes introduced this recently ?

jgoppert, 2015-11-02:
It still works. The ros build is tested on every commit. Try "make ros"

killerlayup, 2015-11-02:
@jgoppert  you mean 'rosmake ?

zshn25, 2015-11-04:
Still the same error. 
[image]

Please help!!

killerlayup, 2015-11-04:
@zshn25 
do you have free time? 
we can discuss on gitter.im if you want to

killerlayup, 2015-11-04:
here is my gitter link:
https://gitter.im/killerlayup/jMAV

jonbinney, 2015-11-10:
The basic problem I'm seeing with the ROS build (that gets run when `catkin_make` is called) is that `Firmware/CMakeLists.txt` assumes that CMAKE_SOURCE_DIR is the Firmware directory, but in the case of catkin_make, that isn't the case.

In a catkin workspaces, you typically have:
```
workspace_root_folder/
    src/
       /Firmware
       /Some_other_ROS_repo
       ....
```

Catkin seems to set CMAKE_SOURCE_DIR to `workspace_root_folder/src/`
Changing CMAKE_SOURCE_DIR to CMAKE_CURRENT_SOURCE_DIR gets me a bit further, but then it fails with:
```
CMake Error: File /home/jbinney/ws/px4/src/cmake/posix/apps.h_in does not exist.
CMake Error at Firmware/cmake/posix/px4_impl_posix.cmake:104 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  Firmware/src/firmware/posix/CMakeLists.txt:4 (px4_posix_generate_builtin_commands)

fatal: Not a git repository (or any of the parent directories): .git
-- GIT_DESC = 
CMake Error at Firmware/cmake/common/px4_base.cmake:710 (string):
  string sub-command SUBSTRING requires four arguments.
Call Stack (most recent call first):
  Firmware/CMakeLists.txt:332 (px4_create_git_hash_header)
```

jonbinney, 2015-11-10:
I wasn't too clear in the previous comment; I was referring to chaging line https://github.com/PX4/Firmware/blob/b3bb8adbc5d3498bb95865820ec7a62bbf14c28e/CMakeLists.txt#L156 to replace CMAKE_SOURCE_DIR with CMAKE_CURRENT_SRC_DIR

won13y, 2016-01-03:
catkin_make --cmake-args -DCONFIG=ros_sitl_simple

brandonr49, 2016-02-03:
Has this been fixed?
I just had the same error zshn25 posted.

brandonr49, 2016-02-03:
Nevermind. This was my mistake.

LorenzMeier, 2016-02-03:
We changed the way how things work, so closing this.

closed 2016-02-03 by LorenzMeier