PX4/PX4-Autopilot #3102: UAVCAN firmware server fails to start

mhkabir, 2015-10-31:
This is on latest master : 
```
nsh> uavcan start fw
uavcan: Firmware Server Failed to Start -1

nsh> free
                      total          used          free     largest
Mem:        229760     127344     102416      98640

```

pavel-kirienko, 2015-10-31:
Stay tuned, the fix is coming up soon. #3005

LorenzMeier, 2015-10-31:
@pavel-kirienko That's not the issue. The issue is that you need to run:

uavcan start
uavcan start fw

You can't run the firmware server straight. Would be good to fix that.

mhkabir, 2015-10-31:
Actually no. The normal uavcan process is already running for me (I have
UAVCAN_ENABLE set to 2)

At boot, it runs the main process, but fails to start the firmware server.

LorenzMeier, 2015-10-31:
Then the firmware server is already running. A PR with a fixed error message would be nice.

mhkabir, 2015-10-31:
It used to give the proper "Firmware server is running/not running" message previously, so I think it's fine on that angle.

None of the connected nodes get node IDs allocated either. Pretty sure f/w server isn't running.

LorenzMeier, 2015-10-31:
Can't help you then. You will need to debug the boot sequence of the system. Our test setups are doing fine, and what you're trying to do (to set can autostart to 2 and then still start the fw server manually) makes no sense. Try to do it in a more structured fashion, carefully observing the outputs.

mhkabir, 2015-10-31:
> to set can autostart to 2 and then still start the fw server manually) makes no sense
Not really. I checked that the autostart wasn't working. At which point I got in via NSh and tried a manual start.

I'll inspect the bootlogs.

mhkabir, 2015-11-01:
Ok, bootlogs and looking at the code got me all the way. The USB console wasn't printing the full error.

I found that the firmware server was failing to start since it couldn't create a directory on the SD card to copy the ROMFS firmware to. Then I had a look at the SD card, which *already* had the "fw" directory created on it. Some more debugging led me to the fact that the SD had bad blocks :P I tried, and it consistently fails read/write tests quite randomly on my PC.

Installed a new SD card, and we're all good! :)

Thanks Lorenz/Pavel :+1: I can now finally proceed to test #3005

closed 2015-11-01 by mhkabir

kuiskas, 2019-02-06:
Hi,
sorry to reopen this. I was trying the uavcan fw in pixhawk 4 and it does not start. Here the warning.
WARN  [uavcan] CentralizedServer init: -13
WARN  [uavcan] Node init failed: -13
ERROR [uavcan] Firmware Server Failed to Start -13

I do not know if this is only in my case or more people have the same problem. 
On startup it already fail to init, tough standard uavcan messaging works.

dagar, 2019-02-06:
@kuiskas please provide more details about your setup. Which flight controller? How is it configured? Can you check a few things?

    top
    free
    uavcan status

kuiskas, 2019-02-06:
I am just testing uavcan in a pixhawk 4 (px4_fmu-v5_multicopter). Just connected to a zubax babel. 
PX4 Configuration: just standard f450 quadrocopter frame 3S ... 
I just did a pull from master today and I just enabled Uavcan as Sensor and ESC (3).
commit: 3959180a2eec82310f92eab8bcb76f41f7cec16d   
There is a SD card. 
Btw I am getting a warn: uavcan low on stack! (292 bytes left). Maybe there is the problem?
**TOP:**
PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                 1052613 63.190   436/  748   0 (  0)  READY  3
   1 hpwork                      14968  0.980   672/ 1780 249 (249)  READY 13
   2 lpwork                       2804  0.178   880/ 1780  50 ( 50)  w:sig  9
   3 init                          946  0.000  2008/ 2484 100 (100)  w:sem  3
 445 top                           105  1.960  1256/ 1684 255 (255)  RUN    3
  15 dataman                        56  0.000   752/ 1180  90 ( 90)  w:sem  4
 443 log_writer_file                 0  0.000   376/ 1148  60 ( 60)  w:sem 34
 144 sensors                     43407  2.673  1184/ 1964 249 (249)  READY 16
 146 commander                   13057  0.802  1568/ 3212 140 (140)  READY 31
 147 commander_low_prio             31  0.000   576/ 2996  50 ( 50)  w:sem 31
 152 uavcan                     107033  6.684  2088/ 2380 252 (252)  READY 11
 158 mavlink_if0                 68010  4.188  1616/ 2532 100 (100)  READY 33
 159 mavlink_rcv_if0              5388  0.356  1240/ 2836 175 (175)  READY 33
 233 gps                          1179  0.089  1080/ 1516 220 (220)  w:sem  5
 264 mavlink_if1                 22691  1.426  1576/ 2484 100 (100)  READY 31
 266 mavlink_rcv_if1              6138  0.356  1120/ 2836 175 (175)  READY 31
 292 px4io                       36612  2.228   880/ 1484 252 (252)  w:sem 11
 308 fmu                          7676  0.445   888/ 1324 252 (252)  READY  7
 396 ekf2                       125240  8.110  4544/ 6572 250 (250)  w:sem 20
 401 mc_att_control              59528  3.832   968/ 1660 251 (251)  READY 18
 406 mc_pos_control              12778  0.713  1040/ 1860 250 (250)  w:sem 11
 410 navigator                    1557  0.089   904/ 1764 105 (105)  w:sem 16
 441 logger                      11933  0.713  1312/ 3540 245 (245)  w:sem 34

Processes: 23 total, 12 running, 11 sleeping, max FDs: 54
CPU usage: 35.83% tasks, 0.98% sched, 63.19% idle
DMA Memory: 5120 total, 1024 used 1536 peak
Uptime: 1611.089s total, 1052.614s idle

**Free:**
             total       used       free    largest
Umem:       485504     207120     278384     275488

**Uavcan status:**
Pool allocator status:
        Capacity hard/soft: 500/250 blocks
        Reserved:  88 blocks
        Allocated: 39 blocks
UAVCAN node status:
        Internal failures: 0
        Transfer errors:   0
        RX transfers:      1745
        TX transfers:      219457
CAN1 status:
        HW errors: 1576994
        IO errors: 1598557
        RX frames: 0
        TX frames: 196153
CAN2 status:
        HW errors: 656
        IO errors: 660
        RX frames: 1745
        TX frames: 217747
ESC actuators control groups: sub: 9 / req: 9 / fds: 4
ESC mixer: OK
INFO  [uavcan] ESC output:
-1000 -679 -1000 -680 0 0
 esc_status_s

        counter: 17148
        esc_count: 0
        esc_connectiontype: 4
[... 100 lines cut ...]
        esc_address: 0
        esc_version: 0
        esc_state: 0
        esc_vendor: 0
Sensor 'gnss':
RX errors: 0, using old Fix: 1, receiver node id: N/A

Sensor 'mag':
devname: /dev/mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Sensor 'baro':
devname: /dev/baro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
channel 4: empty

Online nodes (Node ID, Health, Mode):
        127 OK         OPERAT

pavel-kirienko, 2019-02-07:
Error -13 means `ErrInvalidConfiguration`: https://github.com/UAVCAN/libuavcan/blob/ee272fd48b614da92d8532527d61ebca55c561d2/libuavcan/include/uavcan/error.hpp#L37

It may be caused by a filesystem path being too long -- either firmware cache directory or the storage backend directory:

- https://github.com/UAVCAN/libuavcan/blob/dfcdf22eda16ff06847976fd6c7f40671fc92eb5/libuavcan_drivers/posix/include/uavcan_posix/firmware_version_checker.hpp#L412
- https://github.com/UAVCAN/libuavcan/blob/67da0aae7dd4eedae7905ab98941592cebab4723/libuavcan_drivers/posix/include/uavcan_posix/dynamic_node_id_server/file_storage_backend.hpp#L177

Neither has any relation to the current issue.

kuiskas, 2019-02-08:
Ok I will look at it and post a new issue if it is necessary. 
thanks for the answer.

twaegemn, 2021-07-26:
Does anyone have an update on this? I'm trying to get the hereflow to work but I'm getting the same error -13 as described above. I just connected the hereflow to can1/2 and set UAVCAN_ENABLE=2.

Dmesg is showing:
WARN [uavcan] CentralizedServer init: -13
WARN [uavcan] Node init failed: -13
ERROR [uavcan] Firmware Server Failed to Start -13

I checked for SD card issues and reformatted to be sure, but nothing helped.

top
` PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                  189574 74.456   276/  512   0 (  0)  READY  0
   1 hpwork                          0  0.000   344/ 1260 249 (249)  w:sig  0
   2 lpwork                        100  0.000  1112/ 1612  50 ( 50)  w:sig  0
 492 mavlink_rcv_if0               391  0.181  2816/ 4068 175 (175)  w:sem  6
   4 wq:manager                      0  0.000   376/ 1252 255 (255)  w:sem  3
 490 log_writer_file                 0  0.000   368/ 1164  60 ( 60)  w:sem  3
  17 dataman                        26  0.000   760/ 1204  90 ( 90)  w:sem  4
  19 wq:lp_default                  63  0.090   812/ 1700 205 (205)  w:sem  3
  29 wq:hp_default                2044  0.815  1028/ 1900 240 (240)  w:sem  3
 140 wq:SPI1                      8206  3.260  1492/ 2332 253 (253)  w:sem  3
 143 wq:SPI4                     15107  5.978  1556/ 2332 250 (250)  w:sem  3
 198 wq:nav_and_controllers      12815  5.072  5296/ 7196 241 (241)  w:sem  3
 199 wq:rate_ctrl                 4667  1.902  1128/ 1660 255 (255)  w:sem  3
 205 commander                    1406  0.543  1448/ 3212 140 (140)  w:sig  6
 206 commander_low_prio              2  0.000   692/ 2996  50 ( 50)  w:sem  6
 215 wq:uavcan                    1713  0.724  1736/ 2996 239 (239)  w:sem  3
 220 mavlink_if0                  6950  2.989  1888/ 2572 100 (100)  READY  6
 276 gps                            66  0.090  1016/ 1676 205 (205)  w:sem  4
 308 mavlink_if1                  1388  0.543  1840/ 2484 100 (100)  w:sig  4
 309 mavlink_rcv_if1               358  0.090  2816/ 4068 175 (175)  w:sem  4
 331 px4io                        3690  1.449   980/ 1484 237 (237)  w:sem  4
 447 wq:attitude_ctrl              964  0.452  1236/ 1668 242 (242)  w:sem  3
 466 navigator                      97  0.090   936/ 1764 105 (105)  w:sem  5
 487 logger                        414  0.181  2128/ 3644 230 (230)  w:sem  3
 493 mavlink_shell                   0  0.000   784/ 2020 100 (100)  w:sem  3
 495 top                            85  0.362  1480/ 2028 239 (239)  RUN    3

Processes: 26 total, 3 running, 23 sleeping, max FDs: 15
CPU usage: 24.82% tasks, 0.72% sched, 74.46% idle
DMA Memory: 5120 total, 1024 used 1536 peak
Uptime: 253.750s total, 189.574s idle`

dagar, 2021-07-26:
@twaegemn is this PX4 v1.12? Which board?

I haven't tried the Here flow, but I've been heavily using the ARK flow with current PX4 (v1.12 or master) successfully.

twaegemn, 2021-07-27:
I have tried both the orange and black cube (profiCNC) using PX4 Pro Firmware v1.12 using the normal and mini carrier boards.

twaegemn, 2021-07-29:
@kuiskas were you able to get this working? I'm getting the same response -13 on all pixhawks I have.