PX4/PX4-Autopilot #3111: SITL: Error in VDev management

tumbili, 2015-11-02:
I randomly get a MAG0 self test failure and tracked this down to an issue with VDev management.
I've put a printf here
https://github.com/PX4/Firmware/blob/master/src/drivers/device/vdev_posix.cpp#L194

printing out the device name of the mag which should be /dev/sim_mag
Now here is the output I get in the terminal:
```
in here 43.00000
got valid fd 43.00000
address /dev/sim_mag
returning ok
mag ret is 0.00000
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  TONE_SET_ALARM 1
INFO  ToneAlarm::start_tune
INFO  IOCTL DEVIOCGDEVICEID 196608
in here 43.00000
got valid fd 43.00000
address /obj/vehicle_command0
mag ret is -25.00000

```

As you can see the first mag test succeeds because the device name is correct. However, the seconds mag test (from the second preflight check) fails because the device name is wrong.
Something must have got messed up with the device management.

LorenzMeier, 2015-11-02:
@mcharleb A quick look would be great!

tumbili, 2015-11-02:
@mcharleb Seems to be a problem with the device names (VDev->get_devname). I looped over the filemap and tried to print out the device names. I always get a segmentation fault when trying to print out the name of the device with fd=2.

LorenzMeier, 2015-11-02:
Partial fix: https://github.com/PX4/Firmware/pull/3112

LorenzMeier, 2015-11-02:
Ok, two more things:

  * I'm fixing a similar issue in VDev right now
  * Because FDs here are total and not per PX4 task as in NuttX, we can in theory have multiple consumers opening the same file. We might need an instance count and ensure we only tear down the resource when the last thread gave it up.

@mcharleb I'm tempted to fix this within the current architecture as it seems to teach us a lot about architecture for the next generation. I'm not surprised we had all sorts of issues after looking at this a bit.

LorenzMeier, 2015-11-02:
This one needs testing. Mac OS still crashing with this.
https://github.com/PX4/Firmware/pull/3113

LorenzMeier, 2015-11-02:
@mcharleb I'm wondering if the fact that I don't have gc-sections on Mac OS is the fundamental difference here now. For the virtual file system we need to fundamentally handle the instance count (per-thread references). Is that in place and I missed it, or does gc-sections do that implicitly for us for Linux but fails on Mac OS CLANG?

LorenzMeier, 2015-11-02:
Closing this in favour of https://github.com/PX4/Firmware/pull/3064#issuecomment-151282989.

closed 2015-11-06 by LorenzMeier