PX4/PX4-Autopilot #3122: UAVCAN mag - failsafe/calibration issues

mhkabir, 2015-11-03:
# Issue 1:

The UAVCAN mag isn't getting properly switched over to when it comes up later. "Best" equals 1, the UAVCAN mag, but it isn't getting used.

Voter class output right after boot :
```
nsh> attitude_estimator_q status
...
attitude_estimator_q: mag status:
INFO  validator: best: 1, prev best: 0, failsafe: NO (# 0)
INFO  sensor #0, prio: 50
INFO    val:   0.0914, lp:   0.0917 mean dev:  -0.0000 RMS:   0.0010 conf:   1.0000
INFO    val:   0.4205, lp:   0.4205 mean dev:  -0.0000 RMS:   0.0013 conf:   1.0000
INFO    val:   0.3210, lp:   0.3208 mean dev:   0.0000 RMS:   0.0013 conf:   1.0000
INFO  sensor #1, prio: 75
INFO    val:   0.5176, lp:   0.5176 mean dev:  -0.0000 RMS:   0.0002 conf:   1.0000
INFO    val:  -0.2025, lp:  -0.2025 mean dev:  -0.0000 RMS:   0.0002 conf:   1.0000
INFO    val:   0.2537, lp:   0.2537 mean dev:  -0.0000 RMS:   0.0004 conf:   1.0000
INFO  sensor #2, prio: 75
INFO    val:   0.5176, lp:   0.5176 mean dev:  -0.0000 RMS:   0.0002 conf:   1.0000
INFO    val:  -0.2025, lp:  -0.2025 mean dev:  -0.0000 RMS:   0.0002 conf:   1.0000
INFO    val:   0.2537, lp:   0.2537 mean dev:  -0.0000 RMS:   0.0004 conf:   1.0000
```
If I connect via QGC, suddenly I see "SENSOR FAILSAFE: RETURN TO LAND IMMEDIATELY". I'm not sure yet what the correlation is to connecting to QGC.
# Issue 2:

The next issue is that magnetometer calibrations don't stay. After calibration, the system goes to standby, but on reboot, it again fails to start up properly.
Note that there is not proper error message related to a missing calibration or similar, just that ARMING_STATE_INIT to ARMING_STATE_STANDBY is denied if I try to arm it.

@pavel-kirienko @LorenzMeier Help would be appreciated on this one.

pavel-kirienko, 2015-11-03:
I believe it would make sense to first investigate the weird behavior I observed over there https://github.com/PX4/Firmware/issues/2912#issuecomment-146683852, as it could be (most likely is) related to this one.

mhkabir, 2015-11-03:
I just started using the CAN mag/GPS again after quite some time. Looks like the older issues persist (I thought that it had been fixed) and more.

The calibration issue seems something new though. I never faced this previously. Especially since calibration should not depend on the failover, etc, and only the order in which the sensors come up. 

Can you try reproducing the problem on bench?

mhkabir, 2015-11-03:
Ah. This is something. If I stop and restart commander after UAVCAN has come up, system can successfully go to standby.

LorenzMeier, 2015-11-06:
@mhkabir You will need to debug this in your local setup. @tumbili and I are happy to guide you, but we can't fix it for you.

mhkabir, 2015-11-10:
@LorenzMeier @tumbili I'm still trying to track this down. Looks like the "primary sensor" selection has issues.
```
Critical: warning: primary compass not operational
```
Can you point out the areas in the code which is responsible for this?

mhkabir, 2015-11-10:
Found it : https://github.com/PX4/Firmware/blob/5f69857a575565be26bd7876155cfcc56950a0d6/src/modules/commander/PreflightCheck.cpp#L368

The ```device_id``` never gets updated. Added printfs show this: 
```
[15:56:14.822 - COMP:1] Critical: mag check pass for id -1
[15:56:14.872 - COMP:1] Critical: primary compass NOT found, found id = -1
[15:56:14.924 - COMP:1] Critical: mag check pass for id -1
[15:56:14.975 - COMP:1] Critical: primary compass NOT found, found id = -1
```

mhkabir, 2015-11-14:
This is completely fixed by :
1. #3171 
2. #3183 
3. Zubax GNSS update to increase magnetometer data rate (@pavel-kirienko)

These, once done gives us complete and working UAVCAN sensor support (related : https://github.com/PX4/Firmware/issues/3158) just in time for the v1.1.0 release :)

mhkabir, 2015-11-14:
Closing in favour of #2912

closed 2015-11-14 by mhkabir