PX4/PX4-Autopilot #3128: Lockstep / redundancy interface

LorenzMeier, 2015-11-04:
@pavel-kirienko We've started to work on a triple-redundant lockstep interface, and the intention is to use UAVCAN (with two buses) as interface. It would be great to have you involved in that effort. The first step will be to let three Pixhawks exchange their status, and we'd take it iteratively from there.

pavel-kirienko, 2015-11-04:
Sure, I am interested in participation.

UAVCAN defines NodeStatus message for the purpose of exchanging general status information between nodes: http://uavcan.org/Specification/6._Application_level_functions/#node-status-reporting.

LorenzMeier, 2015-11-04:
They need to exchange attitude and position estimates, not a general status.

pavel-kirienko, 2015-11-04:
Then I would propose to define a custom message.

LorenzMeier, 2015-11-04:
It's nothing custom, it's something that is a core part of the protocol. You will want to eventually support multi-vendor flight control setups.

pavel-kirienko, 2015-11-04:
Defining a custom message would be a first step towards that goal. Once there's a Pixhawk-specific working solution, we can include its messages into the standard message set.

priseborough, 2017-04-30:
To enable three pixhawks to operate as a coordinated entities with fail operative behaviour, data needs to be shared a different points. Functionally, data sharing can be visualised as planes that cut between all systems at  specific locations along the processing chain. These planes share sensor, estimator, command and decision data. The use of median select algorithms provide a reliable method of rejecting faulty data that does not rely on models. Maintaining sufficient independence whilst preventing the possibility of a fight between autopilots will require careful consideration at the sensing, estimation, control and decision making layers. 

Data to be shared could include:

- Timing: Enables the autopilot 250Hz update loops to be synchronised. 
- Parameters: With the exception of sensor calibration, we need each autopilot to be using the same parameters.
- External C2 data.
- Internal Sensors - IMU, mag, baro: This will improve estimator and controller consistency between autopilots. IMU data would be corrected, integrated and downsampled before sharing. The position offset of each sensor would need to be shared, but only on startup. Sensor data would be consolidated using a median select algorithm. 
- External Sensors - Magnetometer, GPS, etc. These need to be shared on dual links.
- Estimator States: This would include quaternions, velocity, position and filter status. Estimator data would be consolidated using a median select algorithm.
- Mission Plan:
- Navigation State: There is a method of coordinating the state machine between autopilots where each autopilot publishes an intention to change state and the state change is actioned when a majority agree. If all three autopilots have different intentions, then a priority table is used. This ensures that we cannot get into a situation where different autopilots are in different states.
- Controller setpoints: These will need to be consolidated using a median select algorithm.
- Controller integrals: Controller integrator states will need to be shared and consolidated using a median select algorithm. Although in theory the use of identical set-points and IMU data makes this unnecessary, it is essential that controller integrals  do not diverge.
- Actuator Demands: If autopilot 250Hz loops are synchronised and the attitude/thrust controllers are using the same IMU data, integrator states and set-points, then loops will output identical demands, however sharing and consolidation of actuator demands using a median select algorithm would guarantee consistent outputs between autopilots if these conditions were not met.

closed 2018-02-22 by PX4BuildBot

REGATTE, 2022-02-28:
Any updates on this from anyone???