PX4/PX4-Autopilot #3130: git version reported by systemcmd "ver" is not necessarily correct

kd0aij, 2015-11-04:


kd0aij, 2015-11-08:
unable to reproduce

kd0aij, 2015-11-08:
git version does not update when e.g. adding a new config file to ROMFS/.../init.d

dagar, 2015-11-08:
I believe the git version is only set at configure time (running cmake). If you just want a quick fix you can touch any CMakeLists.txt and cmake will rerun the configure the next time any part of the build is invoked.

If you want it to always be correct then we need to change this https://github.com/PX4/Firmware/blob/master/cmake/common/px4_base.cmake#L702
to be a custom_command and then add dependencies.

kd0aij, 2015-11-08:
Verified that both "touch CMakeLists.txt && make" is equivalent to "cmake build_px4fmu-v2_default && make", and both regenerate the git hash header. Since that adds less than 2 seconds to my compiles it works for me.

kd0aij, 2015-11-08:
@dagar I've been reading about CMake, and it looks to me as though the new custom_command to generate build_git_version.h, will have to invoke a new CMake script which contains the body of the existing create_git_hash_header function. But I don't see how to specify the dependencies; what we want is to know whether the git hash has changed, but I guess we have to ask git about that. Would it be too crude to make it dependent on a non-existent target?

davids5, 2015-11-11:
In the past this was solved by saving the SHA to a file and diff ing the contents. If it changed then the dependants were built.

kd0aij, 2015-11-12:
@davids5 I don't understand. The problem is that the git SHA is written to a file only at configure time. Was "the past" before the switch to cmake?

davids5, 2015-11-12:
Yes, before the switch to cmake.

AndreasAntener, 2016-02-17:
Hey, could we just to 'rm build_git_version.h' somewhere in the cmake file? We should fix that again..

kd0aij, 2016-02-19:
I think this needs to be addressed by a dev who knows cmake. My questions above have gone unanswered for 3 months, so it won't be me.

LorenzMeier, 2016-05-22:
Fixed.

closed 2016-05-22 by LorenzMeier

kd0aij, 2016-10-17:
@LorenzMeier It would be nice to have a pointer to the fix for things like this.