PX4/PX4-Autopilot #3132: Position lock is (probably) acquired too soon and not checked for anymore until arming

AndreasAntener, 2015-11-05:
Note: Since this happens for us on the runway branch we need to test if this is also a problem on master.

Symptoms after a few test with a hot GPS:
- right after reboot: the mission upload gets acknowledged with the message "no valid position lock"
- when arming: GPS beep, mission can be uploaded successfully now

LorenzMeier, 2015-11-05:
@tumbili I think we generally need to re-check the system status continuously, just not output any new error messages if the error state didn't change. I will look into this, but it would be great if you could drive and supervise this to ensure this gets settled within this week.

LorenzMeier, 2015-11-05:
Quick check didn't turn anything up.

tumbili, 2015-11-06:
@AndreasAntener @LorenzMeier I also don't see the problem here. I remember that we got the "home message" in QGC which means that this was executed
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L869
Further down you see that the home set tune has been played but I think we never got that (is that correct Ada?)

LorenzMeier, 2015-11-06:
@AndreasAntener I'm closing this since I think you got confused by the home reset logic on arming. We set home three times: 1) On the first valid position estimate. 2) on arming and 3) on takeoff, so that the estimate of where to RTL to gets as close as possible to a good location.

Last RTL improvement we need to do:
https://github.com/PX4/Firmware/issues/3137

AndreasAntener, 2015-11-07:
I can see that home is set 3 times, but could it be that the first time is too early for the navigator?

When the mission gets uploaded, the check looks at the flag `_home_position_set` from the navigator. Since we receive the message saying `Not yet ready for mission, no position lock` we know that this flag is not true yet. On arming the home position gets updated once more, like you said, so now the navigator finally has his flag set to true as well, and reports ready for mission. I think it's a race condition because the commander is started before the navigator.

AndreasAntener, 2015-11-07:
BTW: I this is not a "critical" bug IMHO, but the user (like me) doesn't understand why there is "no valid position lock" because he clearly has good GPS reception.

LorenzMeier, 2015-11-17:
Fixed in 557ceca.

closed 2015-11-17 by LorenzMeier