PX4/PX4-Autopilot #3142: Switching from AUTO_MISSION to OFFBOARD causes disarm

Aperture-Laboratories, 2015-11-07:
Arming and launching an auto mission then attempting to intercept and enter offboard mode through the command line (`rosrun mavros mavsys mode -c OFFBOARD`) causes the copter to disarm just before it enters offboard mode as seen in this console output:

```
[ INFO] [1446753694.247621294]: FCU: ARMED by RC
[ INFO] [1446753694.296392836]: FCU: home: 40.8109998, -72.9867036, 29.12
[ INFO] [1446753694.347059669]: FCU: [cmd] arming state: ARMED
[ INFO] [1446753694.396860919]: FCU: [sdlog2] log dir: /fs/microsd/log/2015-11-05
[ INFO] [1446753694.446975002]: FCU: [sdlog2] starting: 20_01_34.px4log
[ERROR] [1446753699.902373420]: FCU: TAKEOFF DETECTED
[ INFO] [1446753699.952230920]: FCU: home: 40.8110003, -72.9867043, 29.17
[ INFO] [1446753702.160364545]: FCU: Flight mode: AUTO_MISSION
[ INFO] [1446753702.213833753]: FCU: offboard mission now running
[ERROR] [1446753702.267086087]: FCU: takeoff to 2.0 meters above home
[ INFO] [1446753702.316828128]: FCU: [mpc] reset pos sp: 0, 0
[ INFO] [1446753702.368789253]: FCU: [mpc] reset alt sp: 0
[ INFO] [1446753716.748527380]: FCU: [cmd] DISARMED by set mode command
[ INFO] [1446753716.798238380]: FCU: Flight mode: OFFBOARD
[ INFO] [1446753716.862074630]: FCU: [mpc] reset pos sp: 0, -1
[ INFO] [1446753716.899864505]: FCU: [mpc] reset alt sp: 2
[ERROR] [1446753716.950984588]: FCU: LANDING DETECTED
[ INFO] [1446753718.033299005]: FCU: [sdlog2] logging stopped
[ERROR] [1446753772.683675345]: FCU: MANUAL CONTROL LOST (at t=842499ms)
[ WARN] [1446753905.532368694]: TM: Clock skew detected (0.023395442 s). Hard syncing clocks.
[ WARN] [1446754097.546026175]: TM: Clock skew detected (0.028595376 s). Hard syncing clocks.
```

. If the same thing is attempted using purely the RC to arm, launch the auto mission then intercept into offboard mode by flipping the RC switch (in my case, channel 7 is mapped to offboard) it works perfectly fine as seen here:

```
[ INFO] [1446754999.159552142]: FCU: Flight mode: MANUAL
[ INFO] [1446755005.400621601]: FCU: ARMED by RC
[ INFO] [1446755005.449630559]: FCU: home: 40.8109993, -72.9867446, 25.51
[ INFO] [1446755005.504745143]: FCU: [cmd] arming state: ARMED
[ INFO] [1446755005.552833268]: FCU: [sdlog2] log dir: /fs/microsd/log/2015-11-05
[ INFO] [1446755005.606949601]: FCU: [sdlog2] starting: 20_23_25.px4log
[ INFO] [1446755011.259402019]: FCU: Flight mode: AUTO_MISSION
[ INFO] [1446755011.273721185]: WP: reached #0
[ERROR] [1446755011.310363394]: FCU: takeoff to 2.0 meters above home
[ INFO] [1446755011.361200227]: FCU: [mpc] reset pos sp: 0, 0
[ INFO] [1446755011.414319977]: FCU: [mpc] reset alt sp: 0
[ERROR] [1446755011.464954310]: FCU: TAKEOFF DETECTED
[ INFO] [1446755011.515931394]: FCU: home: 40.8109993, -72.9867481, 25.49
[ INFO] [1446755019.753143728]: FCU: Flight mode: OFFBOARD
[ INFO] [1446755019.806816520]: FCU: [mpc] reset pos sp: 0, 0
[ INFO] [1446755019.854701478]: FCU: [mpc] reset alt sp: 1
[ERROR] [1446755047.607721815]: FCU: NOT ARMING: Switch to MANUAL mode first.
[ INFO] [1446755051.018336190]: FCU: Flight mode: MANUAL
[ INFO] [1446755052.645962232]: FCU: DISARMED by RC
[ INFO] [1446755052.697809232]: FCU: [cmd] arming state: STANDBY
[ERROR] [1446755052.749816690]: FCU: LANDING DETECTED
[ INFO] [1446755053.917671690]: FCU: [sdlog2] logging stopped
```

Should I skip using the mavros command and try to emulate the RC state on channel 7? Or is this a commander issue as Kabir had suggested?

If any more information/testing is needed, let me know. Thanks.

mhkabir, 2015-11-07:
Offending line here : https://github.com/PX4/Firmware/blob/85ad5a622a0ad4e6accf566ed4ea9357b1287393/src/modules/commander/commander.cpp#L529

I'm not sure why we'd want this? @LorenzMeier @tumbili

LorenzMeier, 2015-11-07:
That's not the offending line. Its the line interpreting the arming state. http://mavlink.org/messages/common#MAV_CMD_DO_SET_MODE

So fix MAVROS to not send a disarm command.

LorenzMeier, 2015-11-07:
Here is the corresponding issue for MAVROS:
https://github.com/mavlink/mavros/issues/423

closed 2015-11-07 by LorenzMeier