PX4/PX4-Autopilot #3146: Check CONFIG_ARCH_BOARD_XXXXX constructs

davids5, 2015-11-07:
The cmake conversion did not propagate the following 
 
ARCHDEFINES += -DCONFIG_ARCH_BOARD_$(CONFIG_BOARD) 
 
While this should be OK as long as we maintain our defconfig files (by hand) to ALWAYS have CONFIG_ARCH_BOARD_XXXXX (make oldconfig will break this, and I have to hand merge) it may be good to visit all the #if CONFIG_ARCH_BOARD_XXXXX constructs and ensure a compile time failure will result if missing.

dagar, 2017-04-11:
Does this still need to be fixed?

davids5, 2017-04-11:
This was fixed in the PR that expunged the #if CONFIG_ARCH_BOARD from the code base. I also added a tool to preserver the CONFIG_ARCH_BOARD etal through oldconfig and menuconfig for board configs not know to NuttX.

The remaining are 
[image]
[image]

Which all look OK for NuttX

I do not see 
CONFIG_ARCH_BOARD_RPI
CONFIG_ARCH_BOARD_SIM
set - If you can validate those we can close this.

davids5, 2017-09-29:
Now that most all the CONFIG_ARCH_BOARD_XXXXX are gone! This is no longer an issue on 1.7

closed 2017-09-29 by davids5