PX4/PX4-Autopilot #3147: SITL broken on master

jgoppert, 2015-11-07:
I have been unable to track it down yet, but somewhere in the following patch, we broke SITL

https://github.com/PX4/Firmware/compare/4345d8...b3554f

@tumbili, @LorenzMeier

jgoppert, 2015-11-07:
I think this looks suspicious: https://github.com/PX4/Firmware/compare/4345d8...b3554f#diff-0da809c04c1d76a5907b7198828781edL825

The low pass filter is being removed.

jgoppert, 2015-11-07:
I think I found it:

```c++
		/* Attitude rates for control state */
+		ctrl_state.roll_rate = _lp_roll_rate.apply(_rates(0));
+
+		ctrl_state.pitch_rate = _lp_pitch_rate.apply(_rates(1));
+
+		ctrl_state.roll_rate = _rates(2);  <------------- OOOPS
```

+1 for getting SITL running for travis.

kd0aij, 2015-11-07:
Looks like that explains why I broke a few props today :)

LorenzMeier, 2015-11-07:
Ugh, this was clearly a dev process fail we will debrief.

@kd0aij Would you mind re-checking if its fixed now?

LorenzMeier, 2015-11-07:
If its not, beta holds a super recent but stable state.

kd0aij, 2015-11-07:
beta flies OK. Will test master next.
Master is now OK on the QAV250.

closed 2015-11-07 by jgoppert