PX4/PX4-Autopilot #3149: Add gzserver to Travis

LorenzMeier, 2015-11-07:


dagar, 2016-02-28:
Do you still want this?

LorenzMeier, 2016-02-28:
Absolutely! We want to run auto-missions, either in Jenkins or Travis. A DroneKit script went into master which allows to command it and we just need the shell script architecture and CI system install routines now. Then we could do SITL flights on every commit.

dagar, 2016-02-28:
Having used gazebo twice now I'm fairly confident travis-ci is not up to the task. What's the sitl01.dronetest.io hardware like?
If you install all the packages I can do the rest of the setup through the web interface.

AndreasAntener, 2016-02-28:
@dagar 
sitl01.dronetest.io currently runs the multiplatform sim with ROS + Gazebo inside a Docker container. I think it can handle Gazebo + PX4.

Are you familiar with Docker? If we create a new Docker container for POSIX SITL we don't have to install packages :)
https://github.com/PX4/containers

dagar, 2016-02-28:
My vague understanding of docker was that it solved a problem I managed to avoid by just sticking with the same flavour and version of linux everywhere. Maybe it's time to take a closer look.

AndreasAntener, 2016-02-28:
;) I can help getting you started by adding a new container and give you access to Jenkins so you can have a look at how it's instrumented (it's not the perfect setup but functional).

AndreasAntener, 2016-02-28:
@dagar : https://github.com/PX4/containers/pull/19

AndreasAntener, 2016-03-06:
@dagar this is the branch with the updated tests and test run scripts: https://github.com/PX4/Firmware/tree/posix_sitl_ci
I added a TODO list in this issues description from what I think is still open. It would be awesome if we can get this going soon.

dagar, 2016-03-06:
Opening a pull request to test jenkins.

AndreasAntener, 2016-04-20:
POSIX SITL integration testing is merged to master, all PRs need to rebase. The old multiplatform ROS build is still functional but will not be tested anymore.
As soon as I have the dev guide update I'll close this and open new issues for the rest of the todos above.

AndreasAntener, 2016-04-23:
Updated docs, added remaining issues.

closed 2016-04-23 by AndreasAntener