PX4/PX4-Autopilot #3158: Verify UAVCAN

LorenzMeier, 2015-11-09:
@devbharat @boosfelm Here are the things that need to be verified:

1) That the compiler you have produces valid binaries for boot loader and application
2) That you have *removed* the capacitors Ben indicated
3) That firmware upgrade and enumeration work with QGC master (there is also now a checkbox to enable the bus)

With that we should have the confidence that things are operational.

boosfelm, 2015-11-19:
gcc_4_8_3 produces valid binaries for boot loader and application. gcc_4_7_4 fails. We will cross validate more versions in the next days. same with firmware upgrade and enumeration with QGC master.
caps and resistors are removed.

LorenzMeier, 2015-12-02:
Ok, we fixed this. It was a bad linker script plus differences in how the different boot loaders interpreted the param storage flash space.

closed 2015-12-02 by LorenzMeier