PX4/PX4-Autopilot #3160: Parameter saving logic in UAVCAN bridge is flawed

pavel-kirienko, 2015-11-09:
@bendyer Check this out: https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/uavcan_servers.cpp#L524-L527:

```c++
				switch (command_id) {
				case 1: {
						// Param save request
						_param_save_opcode = uavcan::protocol::param::ExecuteOpcode::Request::OPCODE_SAVE;
						param_opcode(get_next_dirty_node_id(1));
						break;
					}
				case 2: {
						// Command is a param erase request -- apply it to all active nodes by setting the dirty bit
						_param_save_opcode = uavcan::protocol::param::ExecuteOpcode::Request::OPCODE_ERASE;
						for (int i = 1; i < 128; i = get_next_active_node_id(i)) {
							set_node_params_dirty(i);
						}
						param_opcode(get_next_dirty_node_id(1));
						break;
					}
				}
```

In the case 1, the call to `param_opcode()` will fail with an error report to CLI if there's no dirty nodes present. The error report looks like this:

```
uavcan fw srv: UAVCAN command bridge: couldn't send ExecuteOpcode: -2
```

Error code -2 means [`ErrInvalidParam`](https://github.com/UAVCAN/libuavcan/blob/master/libuavcan/include/uavcan/error.hpp#L26), since the server is trying to call the service on node ID 128, which is not valid.

LorenzMeier, 2015-11-09:
@pavel-kirienko Since you already worked it out, why not providing a quick patch?

pavel-kirienko, 2015-11-09:
Because I am not sure I understand Ben's intent correctly. Either he meant to force execution of `OPCODE_SAVE` on all nodes (like in the case of `OPCODE_ERASE` below), or he just forgot to add a check if there are any dirty nodes available.

LorenzMeier, 2015-11-19:
@bendyer I'm aware you're travelling but feedback would be much appreciated, since its blocking quite a bit of potential momentum in the CAN adoption.

bendyer, 2015-11-20:
OK—I force `OPCODE_ERASE` on all nodes because the QGroundControl erase command looks like it's meant to apply to everything. `OPCODE_SAVE` should only be applied to dirty nodes; in my testing QGroundControl only sent it after sending a parameter update, so I have not encountered the issue @pavel-kirienko raised, but it's certainly reasonable to check that there is at least one dirty node before calling `param_opcode(…)` here https://github.com/PX4/Firmware/blob/master/src/modules/uavcan/uavcan_servers.cpp#L527

I'll submit a PR shortly.

closed 2015-11-20 by LorenzMeier