PX4/PX4-Autopilot #3161: [TECS] Full throttle when max 'pitch up'

mswingtra, 2015-11-09:
We have observed this issue again during last flight on Saturday (20151108). We have flown with master bb2e5b7 (although with old state of EKF but that should not have effect on this particular issue), additionally we had cherry-picked https://github.com/PX4/Firmware/pull/2940 (we confirmed that this fix works).
Logging: http://logs.uaventure.com/view/YfCtTv5kYnK6A67PeMy4fF - again, there has been probably some renaming since our last pull as the tecs inner loop is not displayed. 

Below are snaps showing the issue (values have been rescaled): in the first one, sudden max. pitch up command triggered saturation of thrust integral in tecs, leading to saturation of thrust sp. However, in the second graph, the pitch up command is increased steadily, and suddenly TECS_ThrI saturates. Additionally, at that moment altitude setpoint drops or is reset (pic no. 3).

[image]
[image]
[image]

LorenzMeier, 2015-11-09:
Please plot the altitude and the altitude set point and check what the climb out diff param is. I reduced that recently quite a bit, before it was 25 m. If your altitude error is bigger than that it will climb out. We might want to increase the climb out difference again a bit.

mswingtra, 2015-11-09:
Climb out diff param is set to 25m. Throughout whole flight we have not reached altitude error bigger than 9m - and these happened after thrust sp saturated (full session plot: http://logs.uaventure.com/logmuncher/charts/YfCtTv5kYnK6A67PeMy4fF/TECS_Outer_Loop.png) . Requested plots - circle marks point in time when thrust integral saturated.

[image]
[image]

mswingtra, 2015-11-10:
I have gone through the code again and I found [a line](https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L108) responsible for that - there is a const limit for throttle/pitch input above which the climbout mode is actually triggered. So it was rather trivial, sorry for bothering you.

closed 2015-11-10 by mswingtra