PX4/PX4-Autopilot #3163: UAVCAN ESC enumeration must ignore all non-ESC nodes

pavel-kirienko, 2015-11-09:
Currently it sends begin enumeration requests even to nodes that are not ESC:

```
uavcan fw srv: UAVCAN ESC enumeration: sent Begin request
uavcan fw srv: UAVCAN ESC enumeration: begin request for node 125 timed out.
```

Node 125 in the log above is Zubax GNSS, it is not enumerable.

bendyer, 2015-11-20:
What's your preferred method of checking that something is an ESC? Is it anything with an `esc_index`, anything that has published a `uavcan.equipment.esc.Status` in the last X seconds, …?

pavel-kirienko, 2015-11-20:
Both approaches seem fine, I would prefer whatever is easier to implement.

bendyer, 2015-11-20:
OK—I will go with the `esc_index` check, since that's the parameter referenced in the `uavcan.protocol.enumeration.Begin` request, and we can look for it while we're counting node parameters.

LorenzMeier, 2017-06-04:
@pavel-kirienko @eyeam3 Is this still an issue? I presume not?

pavel-kirienko, 2017-06-04:
It is still an issue.

The approach of testing `esc_index` described above should be avoided since there is no convention that would require ESC vendors to adhere to this specific parameter naming (and there shouldn't be such convention, we've discussed this quite some time ago). We should resort to the other approach of subscribing to `uavcan.equipment.esc.Status`. In fact, the subscriber needed for that is already implemented in the ESC actuator bridge class, we just need to add a tiny bit of logic to it.

closed 2018-02-22 by PX4BuildBot